MinnowBoard-org/bugs-and-help #4: Minnowboard Turbot HDMI Problem.

vk4151, 2016-12-31:
Dear All,

I have purchased Minnowboard Turbot a week ago. I connected my device to a monitor through HDMI. I was installing windows embedded 8.1 but all of a sudden HDMI output was stopped working. I found that serial port was working and so, I updated BootLoader to a latest version. Still problem has not solved. can you please help me out in solving this issue ASAP.

Thanks in Advance.

Regards,

Vamsi Krishna

uhofemei, 2017-01-02:
Hi Vamsi,

To determine if this is a hardware or software issue, could you please try and boot some other operating system on your Turbot. Any modern Linux Live CD will boot to desktop and should work. 

If you don't get graphical output there either, I suggest you file an Return Merchandise Authorization with the vendor you purchased the Turbot from and have them replace it for you.

Thanks,
Ulf

vk4151, 2017-01-11:
Hi Ulf,

As you suggested I tried booting Fedora Live and Ubuntu Live cd, but in Serial Port it ended up with this message:

error: no suitable video mode found. 
Booting in blind mode.

can you please suggest ASAP.

Thanks,
Vamsi

uhofemei, 2017-01-11:
Hi Vamsi, 

I suggest you try to return it with the vendor you purchased the device from.

Thanks,
Ulf

closed 2017-01-18 by uhofemei