MinnowBoard-org/bugs-and-help #5: Wifi AcessPoint or Hotspot MinnowBoard MAX Turbot

AVConceptProducts, 2017-01-04:
Hello,

One of the project we're working on is using the MinnowBoard with the Android 5.1.1 image.
We would like to know if the image manages the HotSpot mode with a Wi-Fi chip plugged in USB.

An other motherboard we are working on (Odroid C2) that does not handle the HotSpot mode.
Apparently, the image does not allow it at the moment ([http://forum.odroid.com/viewtopic.php?t=21543&p=143777](url))

Now on the Minnowboard, when I plug any external  Wi-Fi chip, the OS does not detect it. What should I do to remedy this problem? Are there specific Wi-Fi chips compatible with MinnowBoard ?

Looking forward your help, i've been woking on that for a few days already!
Thanks :)

BrianOttaway, 2017-01-14:
Hello. My apologies for the delay in reply. I gather that you used the Android 5.1.1 image released on [01.org](https://01.org/android-ia/downloads/android-5.1.1-lollipop-minnowboard-max-64/32-bit-userspace), release date: 02 Oct, 2015. This correct? 

In my efforts I have been able to confirm that:
- This old 5.1.1 Lollipop release is no longer supported. 
- At time of release, wifi via external dongle WAS validated. (I haven't ID'd the make/model used for validation)
- The connectivity engineering software support team has changed substantially since then. 

I'm afraid this doesn't leave you with a solution, unfortunately. Have you discovered a workaround? I still have a couple emails out for help...long shots, but I will post an update here with what comes.

BrianOttaway, 2017-01-29:
Closing item. I have been unable to identify the wifi dongle that was used for validation of the Android 5.1.1 image released on 01.org. For such an old image, the end response I've gathered is that it isn't supported any longer. I hope you have been able to find a workaround.

closed 2017-01-29 by BrianOttaway