MinnowBoard-org/bugs-and-help #6: HDMI Not working while booting UEFI.

vk4151, 2017-01-16:
Hello,

I am facing an issue that while booting UEFI, HDMI output is not coming, whereas after booting OS HDMI output is coming. Can you please help me out how to fix this issue.

Thanks,
Vamsi Krishna

BrianOttaway, 2017-01-29:
Hello Vamsi. 

Clarifications Requested:
**1) What monitor is connected? Is it 1080 res?** 
**2) Can you identify the board (V1, Max, Turbot).** I presume you are talking about the MinnowBoard Turbot. 
**3) What is the firmware version?**  Instructions at https://www.minnowboard.org/tutorials/updating_your_firmware. Since you don't have screen at UEFI, you might have to connect via USB serial console; see: https://www.minnowboard.org/tutorials/ftdi-serial-connection. Please do update the firmware if you have not already done so. 

@zwei4 may be have seen this issue before...copying him on thread.

BrianOttaway, 2017-02-07:
Contacted Vamsi via direct email to see if the issue is still present.

vk4151, 2017-02-07:
Dear Brian,

Thank you for your support, my problem has solved after changing monitor... But still I couldn't connect to the TV monitors. It was working with TV Monitors but now its not.

Regards 
Vamsi.

BrianOttaway, 2017-02-09:
Good to hear that the monitor was a culprit. Regarding TV monitor...you might try a shorter or higher quality HDMI cable as a quick test. If it is working on a monitor but TV, that is strange, especially both at 1080.

ebrent8, 2017-02-09:
Mine worked fine on my Sony Bravia TV with HDMI input.

BrianOttaway, 2017-02-09:
...and fine on my Samsung 45" LCD (1080) and Vizio 1 year old (1080).

zwei4, 2017-02-09:
Hi Vamsi, 

You mentioned that " It was working with TV Monitors but now its not."  What kind of changes have you made to your board? Such as  you upgraed the UEFI BIOS or changed BIOS Setup?

vk4151, 2017-02-09:
Dear All,

Thanks for your valuable replies. I found the problem with cable. I changed my HDMI cable with High speed cable, Problem got solved.

Thanks and Regards,
Vamsi Krishna.

Mr BrianOttaway Thanks for your support. I guess you can close this thread.

BrianOttaway, 2017-02-09:
Wonderful. Glad to hear it is resolved. Closing this one up. -Brian

closed 2017-02-09 by BrianOttaway