MinnowBoard-org/bugs-and-help #7: "Boot failed" When trying to install ubuntu from USB device.

theRandomBit, 2017-01-25:
I'm trying to install Ubuntu from a USB drive following the getting started instructions on minniwboard.org.  when I select the boot from "UFI USB device" it says "Boot Failed" and goes immediately back to the main bios menu.  I'm using the most recent version of bios/firmware from Intel and tried preparing the USB drive both on a MAC using unetbootin, and on Windows using Rufus per the instructions on ubuntu's site.  Bother the BIOS and the Ubuntu are 64-bit.  I'm using a minnowboard Turbot.

BrianOttaway, 2017-01-29:
Hello. A few clarifications/questions for you:

1) MinnowBoard Turbot powered by 5V, greater than 2.5 Amp (recomm 4A+).
2) Firmware at [v0.94 64-bit](https://firmware.intel.com/projects/minnowboard-max), confirmed by [UEFI ](https://www.minnowboard.org/tutorials/updating_your_firmware)screen. 
3) Ubuntu install media created on USB flash drive. [Instructions.](https://www.minnowboard.org/tutorials/installing-ubuntu-16.04-on-minnowboardmax). This is my guess is that the install media is corrupted or created incorrectly. Look at the instructions on our website; give it a try our way if you can take a moment to do so. 

I am adding @ebrent8 here, as he created the Ubuntu tutorial for Turbot and is doing the same for our future board. 

Let us know....

theRandomBit, 2017-01-29:
Hi,

    1. Yes, I have a MinnowBoard Turbot using a 5v 4 amp power supply.
    2. Yes, fw version string is, MNW2MAX1.X64.0094.R01.1612052239
    3. Yes, I followed those directions exactly on the minnowboard.org site for creating the UFI flash drive.  I tried creating the bootable USB drive using both the Windows and OSX linked instructions.  All have the same result.

BrianOttaway, 2017-01-29:
@zwei4 : any thoughts? 
@theRandomBit : have you tried both USB ports? Are you installing to eMMC on board?

theRandomBit, 2017-01-30:
I'm using the blue 3.0 port.  I think I might have found the issue.  I went back and double checked everything and found the MD5 sum on the image didn't match the sums listed on the Ubuntu site.  Re-downloading and trying again to see if that was the issue.

theRandomBit, 2017-01-30:
That was the problem.  The iso I had downloaded was corrupted.  Re-downloading and checking the MD5 sum matched fixed the issue.

closed 2017-01-30 by theRandomBit

BrianOttaway, 2017-01-30:
Wonderful. Glad to hear you are off and running. Good luck.