MinnowBoard-org/bugs-and-help #9: Case for Turbot + Silverjaw/wifi/ssd

BrianOttaway, 2017-01-29:
As submitted to email list by user "Michael":

> I have been using Minnowboards for a few years with the half case and it has worked fine.
> 
> I am working on a new project where I would like to travel with a 
> Minnowboard and need something a little more protective.  In this 
> instance, I am also using a silver jaw lure to add a wireless + ssd.
> 
> Is there existing cases or ones that can be 3d printed that work well 
> with the silver jaw lure?  I found the "tacklebox" case, but at $99 it 
> is quite expensive and with the metal case probably wouldn't be great 
> for wireless usage.
> 
> Any suggestions?

BrianOttaway, 2017-01-29:
Hello Michael. Netgate designed and produced the [blue metal case ](http://store.netgate.com/MBX/Case.aspx)for Turbot. As you point out in your question, it would be quite difficult to stuff the Silverjaw with wifi/ssd in that case. :) 

As a first step I am flagging this thread to Netgate to see what they think. They have a mechanical engineer on staff that may be able to provide 3d print files if they have them...for the larger X-Y dimension of the Turbot+Silverjaw. (TBD). Either way, you will have to address the issue of wifi attenuation of the case and any EMI/EMC resulting, as the case isn't designed for full shielding. 

Let's see what Netgate thinks and go from there.

BrianOttaway, 2017-02-07:
@gonzopancho : I don't believe Netgate has anything to meet this need, but I thought I'd ping in Github to double check before I close the issue.

paulcox, 2018-11-01:
I designed and built a 3D printed case a long time ago for the MBMax with Silverjaw and a full PCI card extension, check on thingiverse for the design and pictures. [https://www.thingiverse.com/thing:1102506]