MinnowBoard-org/bugs-and-help #13: Active heatsink for Turbot Dual-Core.

BrianOttaway, 2017-01-31:
As submitted by user "leenmarsh" via direct web form:

> How can I get an active heat-sink for my turbot [dual-core], something similar to the one on the quad core model?

lsoika, 2017-02-01:
The fansink is not available except as part of our quad core boards or in OEM volumes as a customization of dual core boards.  It is not currently available in low volume to end users.  If there is sufficient commercial interest in upgrades of dual core boards, we will re-evaluate but this is not something we are currently seeing high interest in.

Regards,
Len Soika 
ADI Engineering

gonzopancho, 2017-02-02:
@lsoika could ADI sell Netgate a quantity of these as "spares" to be used for either replacement or for someone who needs a fan for 2C?

lsoika, 2017-02-02:
I’m sure it wouldn’t be an issue. Let me dig into it further and get back to you with additional information.

Bigshooter, 2017-02-06:
Is the mounting holes in the board in a standard format that can be filled with another fan?

lsoika, 2017-02-06:
They are 59 mm in pitch between the two pins. Common in some northbridge / southbridge heatsinks. You would need less than 37.5 mm height/width of the actual body of the heatsink to not interfere with other board components.

Bigshooter, 2017-02-06:
Do you believe these products would work?

[Passive](https://www.moddiy.com/products/Aavid-Thermalloy-Northbridge-%7B47%7D-Southbridge-Heatsink-%2859mm%29.html)
[Active](https://www.aliexpress.com/store/product/1PC-FREE-SHIPPING-Cooling-Fan-Heatsink-Pure-Copper-Northbridge-Cooler-for-PC-Computer-59mm-3-PIN/738745_1394430168.html)

lsoika, 2017-02-06:
Passive - likely to be fine. 
Active - 1) from what I see it is 40mm x 40mm so it is too large. 
             2) I don't see the fan voltage. It would need to be 5V and not 12V.

Bigshooter, 2017-02-07:
@lsoika Thanks for all your help,  Mark me down as someone trying to build a commercial product, we will use quad core boards when they are available.  But even then, we need to be able to replace fans when they fail.

flyinprogrammer, 2018-03-13:
Semi-related question, has anyone had success finding/using a larger passive heatsink with a *quad* core board or finding a significantly quieter active solution? I was hoping to use the quad board in a teaching setting; however, having 20 of these running while trying to teach is going to be a problem.

Mark-van-der-Pol, 2018-03-13:
I have built stacks of 4 and 5 quad core boards for testing and clusters and the noise is barely noticeable in the open office layout I work in. AC and desktops are at the same level. 
If you experienced a louder noise or whine perhaps the fan on your unit is faulty.
|\\/|ark

tdanowski, 2018-03-13:
We did not investigate anything other than the heatsink which is shipped with the units. The limiting factor for us was the potential height of an add-on lure card directly above the board.

If you are running it at room temperature, you might be able to get away with a dual core passive heatsink. It hasn't been tested - so this is all speculation. It would also depend on how heavily you are taxing the boards. If you aren't crushing graphics/hardware too much, it would likely be okay. The CPU has a relatively high junction temperature limit compared to some other CPUs. Worst case, the board would start throttling to reduce temperature. If that was unsuccessful and temperature kept climbing the processor might shutdown. Obviously, any modifications would void a warranty.

Terri Danowski
Program Manager

__________ Information from ESET Smart Security, version of detection engine 17050 (20180313) __________

The message was checked by ESET Smart Security.

http://www.eset.com

flyinprogrammer, 2018-03-13:
Here's a recording of me turning on my DPDK-2340: https://www.youtube.com/watch?v=1tXv6a317HQ

Like I say in the video, maybe I'm just crazy. ¯\_(ツ)_/¯