MinnowBoard-org/bugs-and-help #14: No board/mfr ID reporting on custom MB-Max-a2 based board using i210

BrianOttaway, 2017-02-02:
As submitted by Grigory via community email list:

> I built my own board based on MAX 2.
On the second revision of my board a have replaced realtek LAN to intel i210.
By now i have issue.
If i build 94 version of firmware in 64 bit mode everything is okay.
If i build firmware using 32 bit mode there is some problems.
In main BIOS setup screen there is no board information in left upper corner of the screen.
Also when using windows software(AIDA etc) to identify board manufacturer it shows me nothing. No Manufacturer ID no board ID and even no CPU microcode revision.
I did some investigations and found that when firmware runs in 32 bit mode it does not enter MiscSubclassDriverEntryPoint function which calls function AddSmbiosManuCallback which initialize Platform information using mPlatformInfo variable.

> I tried earlier versions of firmware and results is same.

> There is no problems in 64 or 32 bit mode on previous board with realtek LAN. This is only difference.

BrianOttaway, 2017-02-02:
@zwei4 may have some insight after Chinese New Year.

grinux, 2017-02-02:
Hello Brian.
Thanks for forwarding.
I have some updates.
I turned off realtek by removing 25Mhz XTAL from original MAX2.
And both 32 and 64 bit UEFI stops showing ID strings.
I think UEFI stops showing ID when it cant find UNDI device.
As we know i210 have only 64bit UNDI and as a result my custom board shows ID only with 64bit EFI.
And when i disabled realtek on Minnow2 it stops showing in both cases because there is no UNDI device at all.
Could someone points me start point to start source editing to resolve this problem?

BrianOttaway, 2017-02-02:
Thx for the update. For this too I think @zwei4 is the right person to ask for input...that is, after CNY. :)

grinux, 2017-02-02:
New update.
Just disable realtek PCI Express in BIOS setup to simulate this problem.

BrianOttaway, 2017-02-03:
from Brian Richardson:

> Here is info on how to get the MinnowBoard Max/Turbot source ...
https://firmware.intel.com/projects/minnowboard-max
https://firmware.intel.com/sites/default/files/MinnowBoard_MAX-Rel_0_94-ReleaseNotes.txt 

>Thanks … br
>Brian Richardson, Senior Technical Marketing Engineer, Intel Software brian.richardson@intel.com -- >@intel_Brian (Twitter & WeChat) https://software.intel.com/en-us/meet-the->developers/evangelists/team/brian-richardson

BrianOttaway, 2017-02-03:
From Mike Wu: 

> Hi Grigory,

>Yes, you are right. I210 NIC doesn't have IA32 version, which means IA32 firmware image doesn't have UNDI driver if using i210. SmBIOS driver depends on UNDI driver to provide NIC information, without it, SmBIOS driver can't been dispatched. 
We marked it (IA32 image on i210 board missed some information) as known issue when doing 0.94 release. See if we can fix it at next release, if you have patch, welcome to submit to us, we are open source firmware project.

>Best Regards
>Mike Wu

BrianOttaway, 2017-02-07:
**From Grigory via email list**
Hi Mike,
Thanks for answer.

Solution(not quite correct but effectively for me):
1. in SmBiosMiscDxe.inf remove gEfiSimpleNetworkProtocolGuid (2 places)

2. in MiscSystemManufacturerFunction.c comment code which use gEfiSimpleNetworkProtocolGuid and add few strings to initialize SerialNumber  and  SerialNumStrLen 
  //
  //Get handle infomation
  //
  /*BufferSize = 0;
  Handles = NULL;
  Status = gBS->LocateHandle (
                  ByProtocol, 
                  &gEfiSimpleNetworkProtocolGuid,
                  NULL,
                  &BufferSize,
                  Handles
                  );

  if (Status == EFI_BUFFER_TOO_SMALL) {
      Handles = AllocateZeroPool(BufferSize);
      if (Handles == NULL) {
          return (EFI_OUT_OF_RESOURCES);
      }
      Status = gBS->LocateHandle(
                      ByProtocol,
                      &gEfiSimpleNetworkProtocolGuid,
                      NULL,
                      &BufferSize,
                      Handles
                      );
 }
                     
  //
  //Get the MAC string
  //
  Status = NetLibGetMacString (
             *Handles,
             NULL,
             &MacStr
             );
  if (EFI_ERROR (Status)) {    
    return Status;
  }
  SerialNumber = MacStr; 
  SerialNumStrLen = StrLen(SerialNumber);
  if (SerialNumStrLen > SMBIOS_STRING_MAX_LENGTH) {
    return EFI_UNSUPPORTED;
  }*/
  TokenToGet = STRING_TOKEN (STR_MISC_SYSTEM_SKU_NUMBER);
  SkuNumber = SmbiosMiscGetString (TokenToGet);
  SkuNumberStrLen = StrLen(SkuNumber);
  SerialNumber = SkuNumber; 
  SerialNumStrLen = SkuNumberStrLen;

3. in MiscBaseBoardManufacturerFunction.c do the same

 //
  //Get handle infomation
  //
  /*BufferSize = 0;
  Handles = NULL;
  Status = gBS->LocateHandle (
                  ByProtocol, 
                  &gEfiSimpleNetworkProtocolGuid,
                  NULL,
                  &BufferSize,
                  Handles
                  );

  if (Status == EFI_BUFFER_TOO_SMALL) {
      Handles = AllocateZeroPool(BufferSize);
      if (Handles == NULL) {
          return (EFI_OUT_OF_RESOURCES);
      }
      Status = gBS->LocateHandle(
                      ByProtocol,
                      &gEfiSimpleNetworkProtocolGuid,
                      NULL,
                      &BufferSize,
                      Handles
                      );
 }
                     
  //
  //Get the MAC string
  //
  Status = NetLibGetMacString (
             *Handles,
             NULL,
             &MacStr
             );
  if (EFI_ERROR (Status)) {    
    return Status;
  }
  SerialNumber = MacStr;    
  SerialNumStrLen = StrLen(SerialNumber);
  if (SerialNumStrLen > SMBIOS_STRING_MAX_LENGTH) {
    return EFI_UNSUPPORTED;
  }
  DEBUG ((EFI_D_ERROR, "MAC Address: %S\n", MacStr)); */
  TokenToGet = STRING_TOKEN (STR_MISC_BASE_BOARD_ASSET_TAG);
  AssertTag = SmbiosMiscGetString (TokenToGet);
  AssertTagStrLen = StrLen(AssertTag);
  SerialNumber = AssertTag;    
  SerialNumStrLen = AssertTagStrLen;

Grigory

BrianOttaway, 2017-02-07:
**From Mike Wu**
>Hi Grigory,

>Thanks for the patch. I am afraid we can’t accept your patch. It is a w/a which delete the right behavior of code for X64 and IA32 on Realtek NIC. We need something to fit all of them. See if we can get something better at next release. Still, thanks for your participation.

BrianOttaway, 2017-02-07:
**From @zwei4 **
>Yes, we currently have to use MAC address as the unique serial number of a board.

BrianOttaway, 2017-02-07:
**From Grigory** Issue Closed. 

>I understand
>My post just points the way.
>I spent two days :)

closed 2017-02-07 by BrianOttaway