MinnowBoard-org/bugs-and-help #15: Download link for Release 0.94 Firmware 0.94 32-Bit Debug does not work.

nubok, 2017-02-05:
On https://firmware.intel.com/projects/minnowboard-max the download link for Release 0.94 Firmware Download (Dec 2016) for MinnowBoard MAX 0.94 **32-Bit   Debug** does not work. 

Just wanted to mention it.

zwei4, 2017-02-06:
I have forwarded this issue to the web page owner.

BrianRichardsonIntel, 2017-02-06:
The link should work now.
https://firmware.intel.com/sites/default/files/MNW2MAX1.I32.0094.D01.161205.zip

Thanks … br
---
Brian Richardson, Senior Technical Marketing Engineer, Intel Software
brian.richardson@intel.com<mailto:brian.richardson@intel.com> -- @intel_Brian (Twitter & WeChat)
https://software.intel.com/en-us/meet-the-developers/evangelists/team/brian-richardson

BrianOttaway, 2017-02-07:
@nubok . Thanks for pointing this out. Since the issue is closed I will go ahead and close it. Please do stay in touch with your MinnowBoard experience. Happy to help.

closed 2017-02-07 by BrianOttaway