MinnowBoard-org/bugs-and-help #16: RMA contact

benoitmasson, 2017-02-06:
Hello,

The email address rma@minnowboard.org (given by the seller I bought a minnowboard from, and seen on http://wiki.minnowboard.org/Warranty), does not work (`550 Recipient not found` error).

The address contact@minnowboard.org does not work either, so I'm wondering, what is the procedure to request an RMA now?

Thank you

BrianOttaway, 2017-02-07:
Hello Benoit Masson. Can you tell me where you purchased your board? Just to get a head start :)..have you performed any modifications to the board or exposed it to over-voltage? Any other details?

benoitmasson, 2017-02-07:
I purchased it at www.mouser.com, I can send you the invoice if needed.

I plugged it in for the first time a couple of days after receiving it, when I got compatible 5V-4A power supply, CR1225 battery and micro SD card, and it didn't (seem to?) start... Only the power LED gets lit, the board does not output anything (HDMI-USB-Ethernet outputs tested).

Thanks for your help

BrianOttaway, 2017-02-09:
1. I pinged ADI about the thread. Stay tuned. 
2. To clarify your issue. You are seeing D1 LED light up, which means the board has power. Yet, D2 LED is never lighting up? Correct? I'm not making any conclusions here, but D2 LED not coming on usually ends up meaning that the board was fried by an over-voltage power supply (>5V)--see [Power-On Tutorial](https://www.minnowboard.org/tutorials/powering-on-minnowboardturbot). You might double check the power supplies on your desk as it is quite easy to get the 5V mixed up with a 12V.

MinnowBoardHW, 2017-02-09:
Hi Benoit- I just ran a few tests here and I agree with Brian's assessment- if, after pressing the power-button, D2 does not illuminate then board will likely need to be replaced.  It could've been damaged in transit or electrostatic-discharge.  Here are a few things you might check before returning the board though:

1. Disconnect everything, including the the battery & microSD. Re-connect the power-supply and press the power-button to see if D2 illuminates.
2. Confirm that no metal objects are contacting the board, possibly causing an electrical short.

benoitmasson, 2017-02-09:
Hello,

Thanks for your replies. I find it hard to believe that this is due to over-voltage, since I was aware of this 5V limit and was very careful about it, but I'm not the only person in the office... I'll investigate about that.

As for the customer support, Mouser told me in the meantime that they were taking care of the situation, which will be easier for me since they have an office in France. I guess I don't need an RMA anymore then, so I'm now closing the issue.

Thank you again for the kind help.

closed 2017-02-09 by benoitmasson

BrianOttaway, 2017-02-09:
Good. Kudos to Mouser.