MinnowBoard-org/bugs-and-help #17: Windows Embedded 8.1 Standard Not installing in Minnowboard Turbot on SDCARD.

vk4151, 2017-02-09:
Hi,

I tried installing Windows 8.1 Embedded Standard 64-Bit in Minnowboard Turbot on SDCARD. Installation gets started but at the stage "Installing Features and Updates" is never ending. And So, I tried installing Windows 8 Professional 32-Bit, Installation finished smoothly without any struck up. 

As it isrecommended to make SDCARD as NON-REMOVABLE media in Firmware Settings, I Made it same. 

I would like to know, Is there any other configurations need to be changed for installing Windows Embedded 8.1 standard on SDCARD.

Thanks and Regards,
Vamsi Krishna

zwei4, 2017-02-09:
We have tested below win 8.1 with SD card. There is no special requirement. 

 Windows8.1 64bit ISO
      9600.17050.WINBLUE_REFRESH.140317-1640_X64FRE_EMBEDDEDINDUSTRY_EN-US-IR3_CEI_X64FRE_EN-US_DV9.ISO

BrianOttaway, 2017-02-09:
See the firmware tutorial link below. What is your firmware version?
https://minnowboard.org/tutorials/updating_your_firmware

vk4151, 2017-02-09:
Hello Brian,
I am using latest version of Firmware 0.94.. 

Thanks and Regards
Vamsi Krishna

vk4151, 2017-02-11:
Hello All,

I tried installing Windows Embedded Standard 8.1 both 64-Bit and 32-Bit. But No luck, I couldnt instally any flavour of Windows Embedded Standard 8.1. 

Can anyone please help me out to sort out the issue.

Thanks and Regards 

Vamsi Kirshna

vk4151, 2017-02-13:
Hello All,

Problem was solved after Deploying Minimal Image from Image Configuration Editor of Windows Embedded Standard Toolkit. I could be able install both the flavours.

Issue can be closed.

Thanks and Regards
Vamsi Krishna

closed 2017-02-13 by vk4151