MinnowBoard-org/bugs-and-help #18: Boot Graphics Resource Table (BGRT) for Logo Insertion in Minnowboard

vk4151, 2017-02-13:
Hi,

I installed Windows Embedded Standard 8 on Minnowboard Turbot with Unbranded Boot screens. I wanted to Replace the Custom Boot Logo, but unfortunately Windows Embedded Standard doesn't support to Replace Logo, instead we can use Branded or Unbranded(Blank) Logo only. 

According to the MSDN from following link: 
https://msdn.microsoft.com/en-us/library/jj963199%28v=winembedded.81%29.aspx?f=255&MSPPError=-2147217396

Boot Logo can be replaced by Custom Logo only through modifying the Boot Graphics Resource Table (BGRT) on a device that uses UEFI as the firmware interface.

So I request you to please provide information regarding BGRT for Minnowboard Turbot Firmware for Custom Logo Insertion.

Thanks and Regards,
Vamsi Krishna

BrianOttaway, 2017-02-13:
Hello Vamsi. @zwei4 or @uhofemei will have better and more complete inputs I'd think, so I'll leave it to them. Until then, you might take a look at [UEFI.org specs](http://www.uefi.org/sites/default/files/resources/ACPI_5_1_Errata_B.PDF), like section 5.2.22 here, to familiarize yourself with the BGRT fields and definitions.

BrianOttaway, 2017-02-28:
@zwei4 : thoughts?

zwei4, 2017-03-01:
We will make OEM logo / BGRT available in next 0.95 release.  

The minnowboard-max open source code of ACPI tables is available at below repository for your reference:
https://github.com/tianocore/edk2-platforms/tree/minnowboard-max-udk2015/Vlv2DeviceRefCodePkg/AcpiTablesPCAT

zwei4, 2017-03-10:
The patch has been committed to repository  https://github.com/tianocore/edk2-platforms.git  

Patch detail:
https://github.com/tianocore/edk2-platforms/commit/f6a9c795c5fddbdc5d74725748437078697a86ad

BrianOttaway, 2017-03-10:
awesome. Thanks @zwei4 . @vk4151: when this patch makes it into v0.95 firmware release please take a look. I'll close the bug after v0.95 release...but feel free to re-open if you have question on the topic.

keerockl, 2017-04-05:
BGRT table is added in v0.95. closing bug.
Please refer to firmware release link in below.
https://firmware.intel.com/projects/minnowboard-max

closed 2017-04-05 by keerockl