MinnowBoard-org/bugs-and-help #20: Booting problem

daniellejeong, 2017-02-14:
Hi,
I'm using minnowboard Turbot with Ubutn 14.04 LTS,
but ran into some problems.

When I power on the device,
it displays the message "boot failed" twice.
Since the booting order is ubuntu->Fedora->EFI Internal Shell,
I assume ubuntu and Fedora are failing.

Then it shows a UEFI interactive shell where it says

UEFI Interactive Shell v2.1
EDK II
UEFI v2.40 (EDK II, 0x00010000)
map: No mapping found.
Press ESC in 1 seconds to skip startup.nsh or any other key to continue.
Shell>

Does anybody know how to fix this?

zwei4, 2017-02-14:
According to the message, there is no storage device detected. 
      map: No mapping found.

What kind of mass storage devices are attached? SATA Hard Disk? USB cd-ROM?

uhofemei, 2017-02-14:
You need to locate the bootx64.efi boot stub. This can be done either from the BIOS boot manager or manually using the EFI shell. Usually this file is located at FS0:\EFI\BOOT\bootx64.efi or FS1:\ with the same path respectively. Using the BIOS boot manager is the preferred method to change & save the boot order.

BrianOttaway, 2017-02-28:
Presuming that @daniellejeong resolved the problem. Closing issue.

closed 2017-02-28 by BrianOttaway

NicholasJohnson9149, 2018-06-25:
I'm having boot issues with a previously working Minnow - It was working 3 weeks ago and has since stopped booting. I finally got to debugging it today and can't figure out why. D1 and D2 have full power and are one, I checked voltages on D1 and D2 and they are in the clear. I've read a few other boot failure and feel this one comes closest to my issue.  When I plug in the MinnowBoard Dual-core Intel® Atom™ E3826 plugged into a mouse and Keyboard with a micro SD card and a USB with ubuntu 18 desktop (Installer) and a 4K monitor and a 5V 4amp Power supply nothing appears o my monitor, no Bios, no boot screen nothing. I've measured the current to find that it's pulling 0.8 amps, and think my next move will be trying to flash new firmware on the board, anyone know of a good tutorial on how to flash new firmware/bios to these boards?  Not this board has the [m2 and wifi ](https://minnowboard.org/add-ons/silverjaw-lure) adapter with a 32Gb M2 SSD.

keerockl, 2018-06-25:
if you have dediprog spi flash tool, please check tutorial in https://minnowboard.org/tutorials/updating-firmware-via-spi-flash-programmer