MinnowBoard-org/bugs-and-help #21: MinnowBoard 3 (MB3) Market Availability

BrianOttaway, 2017-02-16:
We have received several inquiries regarding the market availability date of the planned MinnowBoard 3, based on an Intel Atom x5-3940 SoC. Instead of creating an email list, we will provide availability updates directly to this issue thread. To receive update notifications, click the “Subscribe” button to the right side of the screen. Updates may also appear on our [website](https://minnowboard.org/).
* **Current Status: MinnowBoard 3 is expected to be commercially available in the Fall of 2017**

felipebalbi, 2017-03-29:
NICE!! Hopefully this one will come with both USB Host and USB Peripheral support ;-)

BrianOttaway, 2017-03-29:
@felipebalbi : For sure. USB-C/3.0 OTG.

tdragovich, 2017-06-21:
Has a price range been determined?

BrianOttaway, 2017-06-28:
@tdragovich : A firm price has not been established.

pietrushnic, 2018-02-14:
@BrianOttaway any news ? I'm watching Mouser, but no news there just coming soon message.

thejae, 2018-03-26:
Looks like it has been abandoned...

Mark-van-der-Pol, 2018-03-26:
The MinnowBoard 3 program has encountered some unexpected delays, but has not been abandoned. 
We expect to be be able to announce the adjusted roadmap within a month.
Thanks for your patience.
|\\/|ark.

pietrushnic, 2018-04-17:
@Mark-van-der-Pol who I can contact to get more details about MinnowBoard 3? I believe this platform is the only reasonable replacement for MinnoBoard Turbot and we have some products that base on that platform.

Mark-van-der-Pol, 2018-04-17:
We are working out the final details on timelines and availability this week and expect to be able to announce that next week. 

The form factor will be brand new, and there will be additional options not currently offered. And we think the enhanced flexibility and options will allow MinnowBoard 3 to find its way into many more prototypes and products.

Stay tuned,,,

|\\/|ark.

pietrushnic, 2018-05-18:
@Mark-van-der-Pol my question was not answered. I also can't see any roadmap update that was expected to be provide almost 2 months ago.

Josh0027, 2018-07-18:
No one is replying on my query. I am very interested in this board but everything is all quiet. Is this another vaporware in the making?

keerockl, 2018-07-18:
Hi,
We are currently working on new version of Minnowboard but currently we don’t have fixed date to announce in public yet. Thank you for your patience. Please stay tuned.

Thank you.
Keerock

Josh0027, 2018-09-01:
Almost two months from my last reply and still no news...

naikrovek, 2018-09-01:
I don't understand all the secrecy, and I guess there's a reason that everyone does it.  GOD FORBID there be ANY HYPE **AT ALL**!

Modern product marketing:

| what's really happening  |  what we say is happening |
| ------------- | ------------- |
| thinking about doing it  | "no comment"  |
| started designing  | "no comment"  |
| prototyping  | "no comment"  |
| a rumor leaks  | "no comment"  |
| final design  | "no comment"  |
| product starts selling | "we've worked on this for 120 years"  |
| a user needs help  | "no comment"  |
| a user wants to buy 200000 of these to use in a product | "that's not how we designed it"  |

Josh0027, 2018-10-06:
So now you guys ditched Minnowboard 3 and replaced it with the Minnowboard 3 Module. When are you going to formally announce it?