MinnowBoard-org/bugs-and-help #22: MinnowBoard MAX CPU Temperature

vetonium, 2017-02-20:
I'm using Ubuntu 14.04 on my Minnowboard and using lm-sensors it looks like the CPU0,2 reach a steady-state of ~72C.

Is this the nominal temperature?

Can the heat sink be easily removed and replaced with a heat strap to conduct the heat away?

BrianOttaway, 2017-02-28:
Commenting on SoC temperatures is always difficult as there are so many impacting variables, like your ambient temps and air flows. yet, 72C sounds about right for what you are describing. I'll put on our list of tutorial requests a more in-depth thermals characterization. 

As long as you don't mind taking chances with your warranty, the adhesive-attached heatsink on a Turbot can be removed with some gentle twisting. Your "heat strap'...how many watts can it safely dissipate?...this should be achievable.

-brian

vetonium, 2017-02-28:
Ok, thanks. I removed the heatsink per your instructions, and I'm ordering a heat strap to replace it.

BrianOttaway, 2017-02-28:
Good luck. Keep an eye on your temperatures. :)

closed 2017-02-28 by BrianOttaway