MinnowBoard-org/bugs-and-help #23: MinnowBoard MAX - Electrolitic Capacitors

vetonium, 2017-02-20:
Does the Minnowboard MAX use any Electrolytic Capacitors?

Is there an electrical schematic?

BrianOttaway, 2017-02-28:
To clarify: Are you talking about MinnowBoard Max or the currently shipping [MinnowBoard Turbot](https://www.minnowboard.org/learn-more)? 

All of the board design files are posted in Github, including schematics and BOMs:
- Turbot dual-core currently shipping: https://github.com/MinnowBoard-org/design-files/tree/master/minnowboard-turbot-dual-core-revX205/pcb-rev-f200
- Turbot quad-core due to ship in a week or two: https://github.com/MinnowBoard-org/design-files/tree/master/minnowboard-turbot-quad-core/F400_R302

Note: @lsoika just posted the quad-core BOM last week so it should all be quite fresh.

If I don't hear back, I'll presume you found the collateral.

vetonium, 2017-02-28:
Hello,

I found the BOM for the MinnowBoard MAX. I noticed that the C76,C77,C87,C129 330uF, 2V capacitors are listed with a CCO_number 123-0005035, but the distributor of the part and it's part number is not listed.

Can you provide the distributor and part number for the 330uF, 2V Capacitors, so I can verify that they are safe for my application? I believe these are Tantalum Electrolytic Capacitors, and it's possible that I may need to replace them.

Thank you

vetonium, 2017-02-28:
Thank you, I was able to find the part on your BOM for the Turbot, and I'm assuming it's within family part for the MAX.

CAP, AL Polymer Elect, SM, 330uF, 20%, 2.0V, 7 mOHM
Panasonic
EEF-SX0D331XE

closed 2017-03-17 by Mark-van-der-Pol