MinnowBoard-org/bugs-and-help #24: minnowboard soft-off

ItayDavidi, 2017-02-28:
Hello,

I'm using the Minnoboard MAX, I wanted to restart the board, so I held the reset button for 10 seconds, but then the UEFI shell gone, how can I recover it?

Thank you

zwei4, 2017-03-01:
Could you give more detailed description about the "then the UEFI shell gone"?

zwei4, 2017-03-01:
If it means the board does not boot anything, such as uefi shell or from USB key , then  you could geta serial cable attached  to the board. Then power on the board to see if there is any messages printed out through the serial port.

ItayDavidi, 2017-03-01:
When I connect 6-Wire serial cable and go to Tera Term I get these to messages only:
>>>>SecStartup
>>>>MemoryInit Done
After that it does nothing.

Thanks for your help!

uhofemei, 2017-03-01:
Is that with an HDMI cable connected? Does the output change if you disconnect HDMI?

ItayDavidi, 2017-03-01:
No, the same

zwei4, 2017-03-02:
There is a recovery jumper J7 on the board. You could try it.

ItayDavidi, 2017-03-02:
how can I use it?

zwei4, 2017-03-02:
Plug a jumper cap to J7.

uhofemei, 2017-03-02:
Also, what are the specs of your power supply?

ItayDavidi, 2017-03-02:
5V 2.0A

ItayDavidi, 2017-03-02:
Do I need to wait or do something other than plug the jumper cap to J7? because it remains the same..

uhofemei, 2017-03-02:
I suggest the best way to proceed at this point will be to file an RMA with the vendor you purchased the board from and have them replace it for you.

closed 2017-03-13 by uhofemei