MinnowBoard-org/bugs-and-help #25: Minnowboard Turbot Urat problem

chenlihao92, 2017-03-01:
Hello,

I'm using the Minnowboard Turbot , I wanted use to change the **original Serial console** to the CPU comes with **High Speed UART1** or **High Speed UART2** . I guess it is to change the code in the serial port register value **0x3f8**, but I do not know what to change, or I think wrong. Can you help me ?

Thank you

uhofemei, 2017-03-01:
Hi, Instructions for using the UART header on the MinnowBoard Turbot are available here: https://firmware.intel.com/sites/default/files/MinnowBoard_MAX_Serial_Console_with_Windows.pdf 

The document was created for the MinnowBoard MAX and will also work for the Turbot. 

Thanks.

chenlihao92, 2017-03-02:
Hello,thank you for your answer, but it can not solve my problem. The serial port in the document I can use, but because of the development needs, I gave up hope to use the serial port, high-speed serial E3845 native 1 or 2 to print DUBEG, need to change the UEFI source code, but has failed to modify, hoping to get help
Thanks.

chenlihao92, 2017-03-02:
[uart.pdf](https://github.com/MinnowBoard-org/bugs-and-help/files/812520/uart.pdf)
I tried Intel Firmware Engine, but also modify the UEFI source code can not solve the problem

chenlihao92, 2017-03-02:
Hi, 
  I'd like to use one of the high speed serial ports on the Minnowboard Max 
as either a terminal or as the debugger connection in EDK2 (UEFI).  Is that 
possible if so can anyone tell me how to get that working. 

Thanks,

zwei4, 2017-03-02:
You first need to initialize High speed UART according to PCI spec, such as assign MMIO resource etc.

Then if you are using SerialPortLib|MdeModulePkg/Library/BaseSerialPortLib16550/BaseSerialPortLib16550.inf ,
You need to configure below PCDs .

  gEfiMdePkgTokenSpaceGuid.PcdUartDefaultBaudRate               |115200
  gEfiMdeModulePkgTokenSpaceGuid.PcdSerialBaudRate              |115200
  gEfiMdeModulePkgTokenSpaceGuid.PcdSerialUseHardwareFlowControl|FALSE
  gEfiMdeModulePkgTokenSpaceGuid.PcdSerialUseMmio               |TRUE
  gEfiMdeModulePkgTokenSpaceGuid.PcdSerialRegisterBase          |0xB0000000
  gEfiMdeModulePkgTokenSpaceGuid.PcdSerialLineControl           |0x03
  gEfiMdeModulePkgTokenSpaceGuid.PcdSerialFifoControl           |0x07
  gEfiMdeModulePkgTokenSpaceGuid.PcdSerialUseHardwareFlowControl|FALSE
  gEfiMdeModulePkgTokenSpaceGuid.PcdSerialDetectCable           |FALSE
  gEfiMdeModulePkgTokenSpaceGuid.PcdSerialRegisterStride        |4
  gEfiMdeModulePkgTokenSpaceGuid.PcdSerialClockRate             |44236800
  gEfiMdeModulePkgTokenSpaceGuid.PcdSerialPciDeviceInfo         |{0x1E, 0x03, 0x00, 0x00, 0xFF}|33

zwei4, 2017-03-02:
After you correctly configured your above SerialPortLib, 
You could add below SerialDxe  driver into your codebase,  which will counsume above SerialPortLib to produce EFI_SERIAL_IO_PROTOCOL .

https://github.com/tianocore/edk2/tree/master/MdeModulePkg/Universal/SerialDxe

chenlihao92, 2017-03-05:
Hi,
Maybe I did not describe it correctly.  When debug info is enabled, by default, UEFI core prints log info to defaul serial console in booting phase.  Here I mean how to redirect these debug log to high speed UART1/2 in booting phase?   Is it configurable?

zwei4, 2017-03-06:
There is no way to make it work without changing code. 

As I have said: You first need to initialize High speed UART1/2  according to PCI spec, such as assign MMIO resource etc.
Then need to configure above PCDs which are used by SerialPortLib|MdeModulePkg/Library/BaseSerialPortLib16550/BaseSerialPortLib16550.inf . 
And make sure you DebugLib is based on this SerialPortLib.

uhofemei, 2017-04-17:
As there has been no further update on this issue, I propose we close the issue. Thanks.

closed 2017-04-17 by keerockl

michaesc, 2017-05-18:
There is no further update, because it seems that UART1 and UART2 are unusable on the MinnowBoard Turbot. That's not a particularly good reason to close the bug, rather it might be better to solve the problem first?

I'm going to start a new report in #40:

https://github.com/MinnowBoard-org/bugs-and-help/issues/40

...because I'm looking for a general solution to use UART1 or UART2 in any way at all (since they seem to be totally broken?)