MinnowBoard-org/bugs-and-help #28: Cannot detect wireless network card

daniellejeong, 2017-03-08:
Hi, 
I've got a lot of helpful comments here and have had some progress,
but I've encountered some new problems and was hoping to get some help :)

I'm a newbie working with boards, so please understand if my questions are silly!

Apparently, my SATA drive seemed to be broken, 
so I installed an ubuntu-14.04 image on SD card and have boot it on Minnowboard Turbot board.

I found a website that gives guidelines and steps for installing ubuntu on Minnowboard Max
(https://01.org/developerjourney/installing-ubuntu-minnowboard-max)
but this was only for ubuntu server.

I wanted to install Ubuntu Desktop, so I made a bootable usb with 
ubuntu-14.04.5-desktop-amd64.iso image.
(I am not sure if it has all the Minnowboard firmwares, but it seemed to work well)

The problem I've encountered is this.
I've attached the Intel Ultimate N 5300 wireless network card to the board,
but the board cannot detect it.
(using commands like iwconfig, ifconfig, lspci...)
It only detects the ethernet port.
Even in the BIOS setting, it detects the ethernet interface, but it doesn't detect any wireless interface.

At first, I thought there was a problem with the network card hardware,
so I tried to change the network card that was working well on another board,
but that chip too was not recognizable,
so I assume it's a configuration problem.

Should I have loaded a different operating system image?

zwei4, 2017-03-09:
I am not sure what's the cause of this wireless problem under Ubuntu Operating System.  
For BIOS stage, it does not include drivers to support wireless network card.

daniellejeong, 2017-03-09:
I see! So I guess I have to look for what went wrong in the operating system configuration!
I will try to update if there's any progress with the problem! Thanks!

gvancuts, 2017-03-09:
My best guess at this stage is that the Ubuntu 14.04.5 desktop image does not include the drivers for such card (which surprises me a bit to be honest). Could you check the `dmesg` content to see what it says? It should typically detect the card but just not able to find and load a driver for it. Or maybe, a firmware is missing (that's another option).

gvancuts, 2017-03-09:
Me again :), just saw your [post](https://ubuntuforums.org/showthread.php?t=2355068) on the Ubuntu forum. What surprises me is that `lspci` does not even list the adapter... so we're not even at a driver issue yet. I haven't played with the MinnowBoard Bios settings in a little while but could it be that some setting is disabling that interface and hence the card is not seen at all by the Linux kernel?

daniellejeong, 2017-03-13:
Hi, thank you gvancuts for your advise!

In the BIOS settings, under Device Manager ->  Network Device List
I can only find one entry in the list which is Realtek ethernet controller,
and no wireless network device!

I looked up keywords like 'wireless' or 'WLAN' in the Minnowboard wiki
http://wiki.minnowboard.org/UEFI

And it seemed like ISCT Configuration under System Setup -> South Cluster Configuration
has something to do with wireless network configuration,
but ISCT Configuration was not on my Minnowboard turbot BIOS menu.

Under System Setup -> South Cluster Configuration ->
I found 'PCI Express Configuration'.
Root Port 1,3,4 is enabled but 2 is disabled.
Does this have something to do with the card device?

Thanks, I really appreciate people taking time to help me!

gvancuts, 2017-03-13:
I _think_ (and hopefully someone can confirm this) that the mPCIe routed through the expansion bus is attached to Root Port 4. Can you confirm how you are installing the WiFi card on your board (and which board you have). Are you already running the latest version of the bios?

uhofemei, 2017-03-13:
Port 2 is disabled in the firmware because it's not used on the MinnowBoard MAX. Enabling it will not make the mPCIe card visible.

daniellejeong, 2017-03-14:
I see!

This is the BIOS version information
>BIOS Information
        Vendor: Intel Corp.
        Version: MNW2MAX1.X64.0090.R01.1601281003
        Release Date: 01/28/2016
        Address: 0xF0000
        Runtime Size: 64 kB
        ROM Size: 64 kB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                8042 keyboard services are supported (int 9h)

The latest version is 0.94, and I assume my BIOS version is 0.90 (from 0090 part).
Is 0.90 also outdated?

The board information (lshw) is
> danielle-minnowboard-turbot-d0-platform
    description: Computer
    product: Minnowboard Turbot D0 PLATFORM (To be filled by O.E.M)
    vendor: ADI
    version: D0
    width: 64 bits
    capabilities: smbios-2.8 dmi-2.8 vsyscall32
    configuration: boot=normal family=IA Notebook sku=To be filled by O.E.M
  *-core
       description: Motherboard
       product: MinnowBoard Turbot
       vendor: ADI
       physical id: 0
       version: REV A
       slot: To be filled by O.E.MEnglish
     *-firmware
          description: BIOS
          vendor: Intel Corp.
          physical id: 0
          version: MNW2MAX1.X64.0090.R01.1601281003
          date: 01/28/2016
          size: 64KiB
          capabilities: pci upgrade shadowing cdboot bootselect edd int9keyboard int14serial int17printer int10video acpi usb zipboot biosbootspecification netboot

I am not sure if I got the meaning of installing a network card right,
but I've inserted the network card in the mPCIe slot which is beside the mSATA slot.
I've checked that the iwlwifi5000.ucode file was in /lib/firmware directory,
and loaded the iwlwifi and iwldvm driver with modprobe command.

> lsmod | grep iwlwifi
iwlwifi                196608  1 iwldvm
cfg80211          561152  3 iwlwifi,mac80211,iwldvm

gvancuts, 2017-03-14:
I don't know if upgrading the firmware could solve this but there is certainly no harm trying this. I saw that the recommendation is also to use a 5V, 3A power supply. What are the specs of the one you are using?

zwei4, 2017-03-16:
by the way, you could try "pci" command under UEFI Shell. It will list out all PCI devices. You could check if your card could be detected by UEFI BIOS.

daniellejeong, 2017-03-16:
The power supply was 5V 4A, but I found what the problem was.
The whole attached Silverjaw Lure expansion board seems to be broken,
so the SATA drive and the network card, both attached to the main board, was not recognizable.
But I really appreciate your kind responses and advises!!! :) Thank you!

closed 2017-03-16 by daniellejeong