MinnowBoard-org/bugs-and-help #30: Noise on LSE Header Causes Interferes with GPS

theRandomBit, 2017-03-15:
Hi,

    I'm trying to interface a GPS via UART1 on a Minnowboard Turbot's LSE connector.  The GPS also draws power from the 3.3v pin.  I've tried multiple GPS devices and none of them work because something about the LSE connector is causing interference and poor reception.  If I take the same GPS devices and attach them via a USB to serial UART they work fine but if any of the pins are connected to the LSE header the device experiences poor SNR and never gets a GPS fix.  

Has anyone experienced this or know ways to mitigate this?

MinnowBoardHW, 2017-03-21:
The 3.3V pin on the LSE header is fed from the main 3.3V power-plane on MinnowBoard which will have some amount of noise from the CPU & related circuits.  The noise levels & frequencies are generally  not an issue for general-purpose prototyping but, for anything wireless, aggressive filtering on the power-supply would be a good idea.

For starters, you might check the GPS module datasheets to see if they have any specific requirements.  If nothing is listed, I would suggest a [pi filter](https://en.wikipedia.org/wiki/Capacitor-input_filter) consisting of a ferrite-bead and two capacitors right at the LSE header.  Make sure you keep the ground connections short too & having more than one can't hurt.

You may also be seeing the effect of near-field coupling into the GPS subsystem, which could easily kill the receiver's sensitivity.  This would explain why relocating the GPS module via the USB UART is helping.  In this case, shielding and/or relocating the GPS module would be the best approach.  Ferrite material is available in sheets and is a good thing to have on-hand for any sort of wireless tinkering.

If you want, reply with a picture of your setup & details on the GPS module that you're using so I can provide some more suggestions.

theRandomBit, 2017-03-22:
It is definitely not near-field.  I was leaving the GPS module in the same spot relative to the board whether it was hooked up through UART1 or the USB-serial.

Putting an LC low pass filter on the Power, TX, and RX lines fixed the issue, the ferrite beads recommended by the GPS manufacture didn't work at all.  I was able to borrow a spectrum analyzer to take a look at the pins using a DC blocker and there is a substantial amount of noise in the 1Ghz - 1.5 Ghz range.  The filter I ended up using covers 70Mhz - 2Ghz, I think the ferrite beads only work in lower frequencies if I'm understanding the data sheets correctly.  If a USB 3.0 device was active that also makes the noise substantially worse.

Thanks for the tips, this was my first run-in with this kind of problem.

closed 2017-03-22 by theRandomBit