MinnowBoard-org/bugs-and-help #31: Where to order just one bare PCB?

antonio-polo, 2017-03-17:
Hello everyone,
The **Relay Lure** posted on the site has two versions: the one with four relays is small enough to mill its PCB on an Othermill; I just completed to assemble one and it's ready for testing. However, the 8-relay version of this lure doesn't fit in the Othermill.  
Any recommendation of a PCB manufacturer who is willing to build ONE bare board from the gerber files without charging a ridiculous amount of money for it?
I've been looking to no avail. One provider is not expensive but warns in advance that "the resulting PCB may have defects" (no thanks); other offers "promotions" that require 4 pieces at a minumum and unreasonable shipping rates for standard surface mail, resulting in having to pay more for the bare PCB than for its components that we already purchased.
Your suggestions are much appreciated. Thanks,

theRandomBit, 2017-03-17:
OSHPark, https://oshpark.com/

BrianOttaway, 2017-03-20:
@theRandomBit has a good suggestion with oshpark. There are many others out there, however, so look around. A few examples: [Advanced Circuits](http://www.4pcb.com/pcb-manufacturing-custom-standard.html), [Sunstone](http://www.sunstone.com/pcb-products/pcb-manufacturing), [SeeedStudio](https://www.seeedstudio.com/fusion_pcb.html). @MinnowBoardHW may have some other opinions. 

**COMMENTARY**: We have talked about, for the mainboards at least, providing the gerber package to a few targeted low-volume suppliers, pre-paying NRE, and negotiating a publicly orderable PCB price...just contact the supplier and order the part number from MB website. Another idea on the list is to setup the board on [Gumstix Gepetto](https://www.gumstix.com/)...to allow simple mods online. Comments welcome.

theRandomBit, 2017-03-20:
SeedStudio has interesting offerings.  I can't understand for the life of me though why anyone would ever deal with Advanced Circuits or Sunstone.  They are so bad.  Their interfaces are horrendous, and their low volume prices are nothing great.

Check out CircuitHub.com or MacroFab.com if you are looking for turnkey PCB Assembly services.  Super easy to use and they don't waste your time.

MinnowBoardHW, 2017-03-21:
Thanks for sharing these options for turn-key PCB fab AND assembly.  This is an area we'd all like to see more progress as it does represent a significant barrier to innovation.  The challenge has always been how tooling & setup fees for fab & assembly get paid for low-volume runs.  Glad to see there are some that are directly tackling this gap though.

antonio-polo, 2017-03-27:
I appreciate your responses. I'll update this thread about this "mission" to get just one PCB done.
Thanks!