MinnowBoard-org/bugs-and-help #32: Debugging BIOS

acckiitvar, 2017-03-23:
Hi. I am going to buy Minnowboard as a stand to practice in UEFI developing. Is it usefull for that purpose? How i can debug my code? I know. that  the board has JTAG interface. What debugging tool and soft I need? Intel System Studio and ITP-XDP 3BR? Or I can use Asset Intertech Arium JTAG Debugger? Or are there any other possibilities to make what i want?
Thanks in advance

BrianRichardsonIntel, 2017-03-23:
You can use the free Intel UDK Debugger over the serial port for source level UEFI debugging.

https://firmware.intel.com/develop/intel-uefi-tools-and-utilities/intel-uefi-development-kit-debugger-tool#overlay-context=develop

https://github.com/tianocore/edk2/tree/master/SourceLevelDebugPkg

Thanks … br
---
Brian Richardson, Senior Technical Marketing Engineer, Intel Software
brian.richardson@intel.com<mailto:brian.richardson@intel.com> -- @intel_Brian (Twitter & WeChat)
https://software.intel.com/en-us/meet-the-developers/evangelists/team/brian-richardson

BrianOttaway, 2017-03-27:
@acckiitvar : does this meet your need? Are you looking to go a little deeper with Intel IDP [Debugger Lure](https://minnowboard.org/debugger-lure) and JTAG?

acckiitvar, 2017-03-27:
I think I'll try both variants. Thank a lot for your help.

closed 2017-03-27 by acckiitvar