MinnowBoard-org/bugs-and-help #33: TPM on Minnowboard

chenho2, 2017-03-29:
Can I get a MinnowBoard TURBO Dual Core board with a TMP chip on it?

Thanks

uhofemei, 2017-03-29:
Information about vendors who sell the MinnowBoard Turbot is available here: https://minnowboard.org/get-a-board

closed 2017-04-04 by uhofemei

chenho2, 2017-04-04:
Yes, I am aware of this fact. However, I contacted Netgate, one of the vendors, and they said to contact this site. I appreciate your response, but I don't understand why it is so hard to get an answer to a straightforward questions. Any other ideas? Has anyone "out there" gotten such a board?

uhofemei, 2017-04-05:
What is a TMP chip? Are you talking about a TPM chip as in Trusted Platform Module? If so, all MinnowBoard Turbot devices have this chip on board.

chenho2, 2017-04-10:
Thank you for the info. Yes, I meant a TPM chip (sorry).

uhofemei, 2017-04-10:
You are welcome :)