MinnowBoard-org/bugs-and-help #34: Admin/User Passwords problem

peralesergio, 2017-04-03:
I'm trying to use Admin/User password configuration without success.
I can introduce them but after save (F10), Commit Changes and reboot/restart, no password is required in order to enter to the UEFI Menu or change Configuration.

I tried with Firmware Images 0.92 and 0.94.

Is there anything I'm doing wrong? What should be the procedure to follow?

uhofemei, 2017-04-03:
What happens if you try to return to the previous BIOS screen using the Esc key? Does that allow you to save the changes using Y/N?

peralesergio, 2017-04-03:
First of all, thanks for your soon answer.
If I do "ESC Key" before F10, there is no question about "Save The Changes using Y/N".
Only if I do "Commit Changes and Exit"

uhofemei, 2017-04-03:
@zwei4 could you intercept BIOS 0.95 validation and check if setting a password is possible with this release?

peralesergio, 2017-04-04:
I tried it with the same result...

13pools, 2017-04-04:
@uhofemei @peralesergio This has always been an issue in the firmware releases - the MinnowBoard platform components and associated UEFI development tooling don't implement this feature (it's also a bit misleading because it fails silently).

I noticed [this patch on the edk2-devel list](https://lists.01.org/pipermail/edk2-devel/2017-February/007090.html) last month - potentially relevant here?

zwei4, 2017-04-05:
Assign to @lushifei

uhofemei, 2017-04-10:
Any update on this issue yet?

peralesergio, 2017-04-10:
I'm trying to merge my edk2 sources with last edk2-git sources and try it.
If any avdance, I will inform you!!Thanks in advance!!!

uhofemei, 2017-05-02:
Hi @peralesergio, any update on this? Thanks :)

zwei4, 2017-05-03:
@lushifei is also looking at this issue from Intel side.

peralesergio, 2017-05-08:
I've been busy with other things. I need to resume work where I left off.
Thanks for your support!!!

zwei4, 2017-05-10:
Patch https://lists.01.org/pipermail/edk2-devel/2017-February/007090.html is on latest edk2 core's trunk. But Minnow Board Max is using edk2 core release package - UDK2015 . So i am not much sure this patch could be applied to Minnow Max source tree. 

@lushifei has implemented a platform level patch for this feature on Minnow.  He will post the patch here. You could integrate it into your existing Minnow Max codebase easily. 

We will not upstream @lushifei's patch to upstream, because it is a platform specific implemenattion. In the future, when patch https://lists.01.org/pipermail/edk2-devel/2017-February/007090.html  is available in a stable version of edk2 core relase pacakge, maybe we will update Minnow Max to that version of edk2 core.

lushifei, 2017-05-10:
You can download source code from Github follow MinnowBaord Max 0.95 firmware releasenote, push this patch to your source code and verify admin password function, while you can use this test firmware image to test admin password on your side.
The attachment is "Admin_Password.zip"
The attachment's password: intelssg
[0001-Vlv2TbltDevicePkg-Add-Admin-Password.zip](https://github.com/MinnowBoard-org/bugs-and-help/files/988460/0001-Vlv2TbltDevicePkg-Add-Admin-Password.zip)

peralesergio, 2017-05-22:
Thank you a lot for the patch.
Apparently I can not patch my sources with your file (probably because of my unexperience)...I'm doing it by hand.
I have problem with .uni files, specially with Vlv2TbltDevicePkg/PlatformSetupDxe/VfrStrings.uni 
When I get it I will let you know.
Thanks again!

BrianOttaway, 2017-09-05:
@peralesergio : Can we close this thread?

BrianOttaway, 2017-09-14:
Closing this thread. Reopen if needed.

closed 2017-09-14 by BrianOttaway