MinnowBoard-org/bugs-and-help #35: Serial port read operation does not unblock when serial port closed

slycote, 2017-04-05:
Hi,

We are running application that use serial port.  OS is Windows IoT 10.0.14393.953 firmware 0.93.

To access serial port we use the windows runtime library  windows.devices.serialcommunication SerialDevice.

Read operation code:

try
{
      // Do the read operation and wait for it to complete.
      auto stream = _serialDevice->InputStream;
      create_task(stream->ReadAsync(_readWriteBuffer, _readWriteBuffer->Capacity, InputStreamOptions::None)).wait();
 }
catch (Exception^ e)
{
    throw ComException(e->HResult);
}

Serial port Close code:

delete _serialDevice;
 _serialDevice = nullptr;

Normaly, read operation should unblock when serial port is closed.  However, this is not the case when running on Minnoboard.  We tried same software on desktop (UWP application) and it work OK.

keerockl, 2017-04-05:
Hi slycote,
from the code above, _serialDevice instance is deleted in Close code, and I don't see any part where _serialDevice instance is created. Are you having exception error when you run this code?
Could you share full code for it? Have you tried running on different OS?

keerockl, 2017-04-17:
Closing issue as there has been no further update on this issue.

closed 2017-04-17 by keerockl