MinnowBoard-org/bugs-and-help #36: Can Not Boot to UEFI Screen

NicholasJohnson9149, 2017-04-10:
Hello, 

I got a minnow board awhile back ad have final got all the accessory i needed to run it this week, I have a 64Gb SSD and a WiFi AC card connected through the breakout board. I'm running off the a 5V 2A power ac/dc adapter. I have this plug into a a 4K monitor, and have all the required connection.for Ethernet, SD card, but I cannot get this device to boot at all, are show a UEFI screen. I got an early release of this board and  would be interested in knowing if I need to install bios because it wont boot to anything?  Can you send me a link to how I might go about doing this.

Mark-van-der-Pol, 2017-04-11:
Take a look at the https://minnowboard.org/tutorials/powering-on-minnowboardturbot tutorial - this walks through the various startup steps and checks. It also links to procedures to update the firmware if that is required.

If you've done all that and it still doesn't work, please provide details of what is on the screen.

The only other thing I can think of is that the specific SD card you have is not compatible. I've heard rumors that 
1) some high capacity SD cards are fakes.
2) some ultra high speed cards are flaky on some minnowBoards.

Good luck! 
|\/|ark

NicholasJohnson9149, 2017-04-11:
Looks like the CPU doesn't get power D2 never turns on, .I will have to ask to have this board replaced..  Thank you for the help.

Mark-van-der-Pol, 2017-04-11:
Sorry to hear that - and before you give up, what happens if there is nothing connected at all, other than the power supply? That is, no monitor, no USB keyboard, nothing on any other connector, no SDCard - just the power jack in the power socket. Does the D2 light up?

|\/|ark

closed 2017-04-17 by keerockl

NicholasJohnson9149, 2018-06-25:
Hello @Mark-van-der-Pol D2 does not light up regardless of what is plugged into the board, the power stage on these boards is not a great design. The voltage on D2 is .8V, it looks like it should be 3.3V.  I did some more exploring and checked out the schematics for the minnowbaord and found that the power stage is not robust enough to handle an over or under voltage spike and has minimal protection. I find that cheap... Intel Nuc's are replaced all my minor boards because they keep failing, I love the idea just not the execution.  I will say that the customer service is amazing, so I appreciate them for trying, and if I have more projects that won't completely fail if the board goes down I'll consider a minor board again.  But right now the reliability is 5/5 failed in the last 3 years.