MinnowBoard-org/bugs-and-help #37: EAGLE or CAD files

marcusbarnet, 2017-04-17:
Hi to all guys,

I would like to ask you if it is possible to have the schematic files (for example in EAGLE or in any other CAD software) of your dual core board (F400_R302). I looked in the folder on github but i only found the PDF schematics. Thank you a lot!

keerockl, 2017-04-17:
Hi marcusbarnet,
Please refer to the git-hub link below. 
https://github.com/MinnowBoard-org/design-files/tree/master/minnowboard-turbot-dual-core/F400_R302

Thank you,
Keerock

marcusbarnet, 2017-04-17:
Thank you Keerock.
I already looked at that link but I have found only the scematic in PDF and the Gerber files.
Unfortunately, I wasn't able to find any CAD file.

keerockl, 2017-04-18:
As I understand, TurbotB_F400_R302_Schematic_Layout_Database.zip has database file and .prj file for CAD.
Can you please let me know if these are what you are looking for?

marcusbarnet, 2017-04-18:
Thank you again for your help, Keerockl.

I've seen that folder, but if I understood it correctly, it seems that files are for Mentor software and they cannot be opened by EAGLE. Since I'm only using EAGLE, I think I won't be able to open any schematics or project, unfortunately.

keerockl, 2017-04-18:
Correct, sorry, but we don't support EAGLE yet.

BrianOttaway, 2017-09-05:
item closed. We will revisit EAGLE CAD conversion at some point. At this time it is not a simple process...unless EAGLE or some brave community member can help out.

closed 2017-09-05 by BrianOttaway