MinnowBoard-org/bugs-and-help #38: how to enable PCIe #1 in software on the Atom 3825 Processor

vchagari, 2017-04-27:
Two mPCIe interfaces are added to the Custom Minnowbooard max board to connect two WiFi modules. One mPCIe interface is connected to the PCIE #1 and another is connected to the PCIE #3 on ATOM 3825 Processor. I could able to see one PCIe bus (PCIE #3) in the OS, using lspci and I am unable to find PCIE #1 interface in the OS. I enabled all the PCIe slots in the BIOS and also included all the necessary kernel modules in the kernel config.  
 
Please tell me if anyone knows how to bring up the PCIe #1 interface, any suggestions/ inputs are highly appreciated.      
 
Details:
 
     Processor: INTEL ATOM
     OS: OpenWrt/ Buildroot
 
     lspci result:
00:00.0 "Class 0600" "8086" "0f00" "8086" "7270"
00:02.0 "Class 0300" "8086" "0f31" "8086" "7270"
00:14.0 "Class 0c03" "8086" "0f35" "8086" "7270"
00:1a.0 "Class 1080" "8086" "0f18" "8086" "7270"
00:1b.0 "Class 0403" "8086" "0f04" "8086" "7270"
00:1c.0 "Class 0604" "8086" "0f48" "8086" "7270"
00:1c.2 "Class 0604" "8086" "0f4c" "8086" "7270"
00:1c.3 "Class 0604" "8086" "0f4e" "8086" "7270"
00:1f.0 "Class 0601" "8086" "0f1c" "8086" "7270"
00:1f.3 "Class 0c05" "8086" "0f12" "8086" "7270"
02:00.0 "Class 0200" "10ec" "8168" "10ec" "8168"
03:00.0 "Class 0280" "168c" "0046" "168c" "cafe"

Thanks
Vamsi

Mark-van-der-Pol, 2017-04-28:
Hi Vamsi,
Thanks for the detailed question, and I still need a bit more, so please indulge me:

Where are the design files for this Custom MinnowBoard MAX, and can you share them? There could be an issue in the design which is the source of this.

And lspci only reports a bus and devices on the bus if there are devices on the bus - an idle bus doesn't show up. So as an experiment, unplug the PCIE #3 and see if lspci now reports nothing for bus three. Then, put that device in slot PCIE #1 and see if anything changes. Finally, put the device that was in #1 into #3 and see what happens. Apologies if you've already tried these combinations, but as you didn't report this I had to ask.

Finally, what version of the BIOS code are you running? The latest is 0.95, and while I am not aware of any recent fixes in this area, it never hurts to stay updated. Well, rarely hurts...

Thanks, |\\/|ark.

vchagari, 2017-05-09:
Hi Mark

Thank you so much for the reply. 

I already tested all those combinations Mark,  my apologies for not mentioning in detail.

When I connect the WiFi module in PCIE#3 it shows up when I do lspci and when I disconnect the WiFi module it doesn't show up when I do lspci. 
Whereas the PCIE #1 interface doesn't show up at all (lspci), I tested it by connecting the WiFi module.

I am using the 0.90 version, I will update it. 

Do I have to manually enable the PCIE#1 interface in the software?. 

Thanks
Vamsi

Mark-van-der-Pol, 2017-05-09:
There is an option in the BIOS to enable and configure the speed for the PCIe busses, but by default these are enabled and set to auto. So possibly, yes, there is that that could be in the way. It is an option in the UEFI Bios, during boot press \<DEL\> to access the UEFI menu. In the System Menu, under the South Cluster, there are PCIe settings.

Let me know if that makes a difference.

|\\/|ark.

vchagari, 2017-05-09:
Hi Mark,

Thank you for the reply. 

I already enabled the PCIe options in the BIOS. This doesn't help to bring up the interface. 

I am also looking at the BIOS src code, to see if anything needs to be done. 

Any inputs and suggestions are appreciated, If you have an experience in the BIOS software (Intel specific) or anyone who worked on it, please let me know.

Thanks
Vamsi

zwei4, 2017-05-10:
We will have a look if the PCEe root port #1 is visiable on Intel Minnow Board Turbot.

vchagari, 2017-05-10:
Hi zwei4,

Thank you for the reply, 

Could you please check and tell me. 

Thanks
Vamsi

lushifei, 2017-05-10:
I am using the 0.90 BIOS with PCI Express Root Port 2 Enable(Device Manager->System Setup->South Cluster Configuration->PCI Express Configuration->PCI Express Root Port 2) and 0.95 BIOS, looks like, the PCIE Root Port #1 is visiable under shell and Yocto on our Intel MinnowBoard Turbot Board. See following detail information:
root@intel-corei7-64:~# lspci
00:00.0 Host bridge: Intel Corporation Atom Processor Z36xxx/Z37xxx Series SoC Transaction Register (rev 11)
00:02.0 VGA compatible controller: Intel Corporation Atom Processor Z36xxx/Z37xxx Series Graphics & Display (rev 11)
00:14.0 USB controller: Intel Corporation Atom Processor Z36xxx/Z37xxx, Celeron N2000 Series USB xHCI (rev 11)
00:1a.0 Encryption controller: Intel Corporation Atom Processor Z36xxx/Z37xxx Series Trusted Execution Engine (rev 11)
00:1b.0 Audio device: Intel Corporation Atom Processor Z36xxx/Z37xxx Series High Definition Audio Controller (rev 11)
00:1c.0 PCI bridge: Intel Corporation Atom Processor E3800 Series PCI Express Root Port 1 (rev 11)
00:1c.1 PCI bridge: Intel Corporation Atom Processor E3800 Series PCI Express Root Port 2 (rev 11)
00:1c.2 PCI bridge: Intel Corporation Atom Processor E3800 Series PCI Express Root Port 3 (rev 11)
00:1c.3 PCI bridge: Intel Corporation Atom Processor E3800 Series PCI Express Root Port 4 (rev 11)
00:1f.0 ISA bridge: Intel Corporation Atom Processor Z36xxx/Z37xxx Series Power Control Unit (rev 11)
00:1f.3 SMBus: Intel Corporation Atom Processor E3800 Series SMBus Controller (rev 11)
03:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 0c)

vchagari, 2017-05-22:
Hi Lushifei,

Thank you for the reply, I already tried enabling the Root Port 2 in the BIOS, I am using 0.90 version.  After saving the BIOS settings and doing a hard/soft reboot, the Root port 2 goes to default settings (disabled), I tried multiple times to cross verify this behavior. 

I also tried updating the BIOS to 0.95 version, I observed that the recent releases (0.95, 0.94, 0.93, 0.92, and 0.91) doesn't work with the Blue Minnow board Turbot, the board doesn't boot up correctly. I am using 64 bit OpenWrt Kernel. 
   
I found a software - **GUI (Intel Firmware Engine)** on the Intel website which can be used to generate the BIOS (https://firmware.intel.com/learn/intel-firmware-engine/downloads). I tried to generate the BIOS using the Firmware Engine by adding the Mini PCI Express slots to the PCIE1 and PCIE3 interfaces, unfortunately, the firmware engine says that it is **unable to determine a firmware module solution**, But when I remove the Mini PCI Express slots from the GUI, the firmware engine will generate the BIOS. Please see the screenshots attached below.  

The firmware engine could able to generate the BIOS when I add PCI Express X1, X16 and X8 slots to the PCIE root ports. It is very strange to me, why does the Firmware Engine fails to create the BIOS when I add the Mini PCI Express slots to the PCIE root ports. 

[image]
[image]

Any help is very much appreciated. 

Thanks
Vamsi

BrianRichardsonIntel, 2017-05-23:
For the issue with Intel Firmware Engine, you can submit your question to https://software.intel.com/en-us/forums/intel-firmware-engine for support.

Thanks … br
---
Brian Richardson, Senior Technical Marketing Engineer, Intel Software
brian.richardson@intel.com<mailto:brian.richardson@intel.com> -- @intel_brian (Twitter & WeChat)
https://software.intel.com/en-us/meet-the-developers/evangelists/team/brian-richardson

[screenshot-pci-express-xx-slots-connected]<https://cloud.githubusercontent.com/assets/10948110/26330260/ed47b6ec-3eff-11e7-9d77-19a13ce43a36.PNG>
[screenshot-mini-pci-express-slots-connected]<https://cloud.githubusercontent.com/assets/10948110/26330265/f13ecad8-3eff-11e7-9ad3-ef55e0adc56c.PNG>

vchagari, 2017-05-23:
Hi Brian 

Thank you for the reply, I submitted the request in the forum.

Thanks
Vamsi

vchagari, 2017-05-23:
Hi Lushifei,

Thank you for the reply, I already tried enabling the Root Port 2 in the BIOS, I am using 0.90 version. After saving the BIOS settings and doing a hard/soft reboot, the Root port 2 goes to default settings (disabled), I tried multiple times to cross verify this behavior. It looks like I can't able to modify the BIOS settings. 

I also tried updating the BIOS to 0.95 version, I observed that the recent releases (0.95, 0.94, 0.93, 0.92, and 0.91) doesn't work with the Blue Minnow board Turbot, the board doesn't boot up correctly. I am using 64 bit OpenWrt Kernel.

Thanks
Vamsi

vchagari, 2017-06-02:
I could able to boot the MinnowBoard with the latest firmware. The board I had before has some hardware issue. 
I could able to see all the PCIe root ports when I do lspci. 

Thank you, everyone.

closed 2017-06-02 by vchagari