MinnowBoard-org/bugs-and-help #39: UEFI Bios 0.95 Spelling and Grammar errors in Help text

Mark-van-der-Pol, 2017-05-09:
I noticed a few errors in the help text of some of the lower nested levels.

System Power and Performance (PnP) Configuration menu, LPC PnP Setting - the help text spells 'diable'

Thermal: Processor Participant - Critical and Passive help texts reads, in part: 'the point in which the processor...'  conventional use is to say 'the point at which the processor...'

Boot: Fast Boot - Help text formatting is off - at least an extra space, and possibly an extra lf/cr.

South Cluster: SATA Drives: Chipset SATA - the help text mentions the 2 black internal ports. There is only one SATA port on MB Turbot. It does happen to be black, though. It is Port 0.

|\\/|ark.

ebrent8, 2017-05-09:
Is this on the website?  If so please send a link.
Also, web bugs go under the web issues.  
https://github.com/MinnowBoard-org/website/issues

This issues list is for board bugs.

ebrent8, 2017-05-09:
@BrianOttaway This is on the board.  Maybe Chris can fix this?

BrianOttaway, 2017-05-09:
Sorry....didn't read this carefully. @keerockl can bring this up with @zwei4 and resolve in next FW release.

zwei4, 2017-05-11:
About SATA，Turbot does have 2 SATA ports . One port is conneected to the black SATA slot, the other is connected to HSE.  The color "black"  and other typos will be fixed.

BrianOttaway, 2017-09-14:
Closing this one...

closed 2017-09-14 by BrianOttaway