MinnowBoard-org/bugs-and-help #40: UART1 and UART2 input and output fails

michaesc, 2017-05-18:
This partially relates to https://github.com/MinnowBoard-org/bugs-and-help/issues/25 where the reporter wanted to redirect boot messages from UART0 to one of the high speed UARTs (UART1 or UART2.)

TEST

I'm trying to use UART1 (for a new sensor carrying lure) and am testing the UART1 so:

minnow# echo 'Hello' >/dev/ttyS1
laptop# cat /dev/ttyS1

PROBLEM

...nothing is printed unless I switch (GND, RX, and TX cable contacts) to UART0:

minnow# echo 'Hithere' >/dev/ttyS0
laptop# cat /dev/ttyS0
Hithere

REPRODUCING

I've reproduced this problem on both UART1 and UART2, connecting CTS and RTS or neither, on Wind River Pulsar8 and Debian and Kali (all up to date), with a default MinnowBoard firmware (AMD64 arch) flashed this month, with either 'Internal UART' or 'SuperIO UART' in UEFI settings, and a variety of baud rates using cu(1) and screen(1) on the laptop. I've set the baud rate of the Minnowboard using:

minnow# stty -F /dev/ttyS1 115200

...and tried killing getty processes as well as passing 'console=ttyS1,115200n8' on the linux root line of grub.cfg. I've also stopped the UART0 systemd service and started the UART1 one and much more.

Computer: MinnowBoard Turbo Dual Core
OS: Wind River Pulsar8 or Kali Linux 2017.1
Cable: Six pin FTDI cable, rather standard
Control: It works with UART0 but not UART1

QUESTION

How can I get input and output from UART1 on the minnowboard?

michaesc, 2017-05-18:
By the way, on the laptop connected to the MinnowBoard with a FTDI cable, I access the UARTs (where UART0 succeeds but UART1 fails) with either cu(1) or screen(1) like this:

laptop$ cu -l /dev/ttyUSB0 -s 115200
laptop$ screen /dev/ttyUSB0 115200

jnavarro7, 2017-05-18:
Hi! 
UART1 is normally mapped to /dev/ttyS4 as mentioned here: http://elinux.org/Minnowboard:MinnowMax

jnavarro7, 2017-05-18:
Unless it is different in Wind River Pulsar Linux(never tried it).  I would start by  looking for /dev/ttyS4 and give it a try. 
laptop$ screen /dev/ttyUSB0 115200 
minnow# echo 'Hithere' >/dev/ttyS0

michaesc, 2017-05-18:
On all OS I've tried (Wind River, Debian, Kali) there have been four devices:

minnow$ ls /dev/ttyS*
ttyS0    ttyS1    ttyS2    ttyS3

...although writing to ttyS3 results in an 'output error'. UART0 clearly corresponds with ttyS0 since when I connect a FTDI cable to UART0 I can communicate correctly with ttyS0.

I see the page you mention, and find it strange that no ttyS4 shows up in my device tree. Is that what you see in your device tree? Maybe the elinux documentation is wrong?

jnavarro7, 2017-05-18:
/dev/ttyS4 is what I see under Ubuntu 14.04 and 16.04.  I used UART1 a couple of days ago to communicate with a  LoRa module and had no problems. 
I will give Pulsar Linux a try in the upcoming days and report back.

michaesc, 2017-05-18:
It could be because I have a stock firmware with UEFI left at default and recent flash of:

MNW2MAX1.X64.0095.R01.1703281124

that I get ttyS(0-3) and no ttyS4. I've solved this problem by spending hours changing a variety of (1) port names, (2) baud rates on both MinnowBoard and Workstation, (3) RX and TX contact connections, (4) TTL cables, and (5) operating systems. There was no way to rule out possible flaws due to lacking or poor MinnowBoard documentation.

There are several thousand such combinations, and I found the one that worked after just 350 tries. It would be nice if better documentation existed to help users avoid unneeded waste. The online specification of UART device name is also wrong, as it's not always ttyS4 for UART1.

closed 2017-05-18 by michaesc

gvamsk, 2017-08-30:
Hi michaesc,

Were you able to find UART 1 and 2 mapping to /dev/ttyS4 or S5 or anything else. 

Are you able to use these serial lines.

Thanks,
Vamsi

michaesc, 2017-08-31:
Hello Vamsi,

Yes, I was able to get more than just the UART0 pins working and I figured out the Pulsar Linux dev tree mapping as well. Thanks for asking.