MinnowBoard-org/bugs-and-help #42: MinnowBoard Turbot doesn't boot up

vchagari, 2017-05-24:
Hi

I used the latest firmware 0.95 version on my MinnowBoard Turbot, It doesn't boot up now. 

It shows the following message and then the board will hang after that.   

" >>>>Start boot option, Press <F2> or <DEL> to enter setup page(5 Sec)..... " 

I tried multiple times with few other versions of the firmware too. I tested the board with 0.94, 0.93, 0.92 and 0.91 versions as well, they didn't work, only the 0.90 version works with my board. 

Can anyone please tell me why the recent versions of the firmware are not working with MinnowBoard Turbot?.

Thanks
Vamsi

BrianOttaway, 2017-06-01:
Vamsi: Is this still an issue? Sorry this reply is late. Your 5V3A PSU should be sufficient, though we do recommend 4A to accommodate higher load setups. 

Turbot boards are used in all firmware validation rigs, so it does not seem likely that there is a true FW issue...but perhaps a conflict with your setup or a board damage/issue. Can you try a simple HDMI connection with high-quality cable, and remove unnecessary connections? Still an issue?

vchagari, 2017-06-02:
Hi Brian
Thank you for the reply.
I have two boards, both showed the similar behavior, I borrowed another Turbot board and flashed the latest firmware, it worked.  

I used 5V and 4A power supply and nothing was connected to the board, expect the UART cable for serial console and USB flash drive to USB 2.0. 

Thanks
Vamsi

closed 2017-06-02 by vchagari

naiyining, 2019-02-18:
i have the same problem.
when first connect the power,sometimes i don't work. and when disconect the power,i will work.
and some times when you wait after about 5-20 second, the board is turn on.
why?