MinnowBoard-org/bugs-and-help #43: MinnowBoard Turbot USB not working

nphamilton, 2017-06-01:
I haven't installed any software onto the board, it is brand new and I was trying to start it up according to the website's tutorial. I entered the shell without issue, but when I tried to type "date" nothing happened. None of the keystrokes are registering. I tried using both USB inputs, but achieved the same result. I also tried turning it off and back on, but also no luck. What should I do?

keerockl, 2017-06-02:
Hi @nphamilton, 
Can you please let me know what is the voltage and amps for the power adaptor?
Also could you check if the keyboard you currently connected to Minnowboard is actually working okay with any other PC that you have?

nphamilton, 2017-06-02:
The power supply is 5V at 4A and the keyboard works because I have connected it to other MinnowBoards without issue.

keerockl, 2017-06-02:
It looks like USB hardware failure on the board from manufacturer. you may want to contact with supplier and request for replacement with a new board.

vchagari, 2017-06-02:
I also faced a similar issue with one of my Minnowboards. Can you please check whether the USB interface is enabled it in the BIOS?, which version of the BIOS firmware the board has?. 
In the latest versions, the USB interface is enabled by default. If it is enabled and still you can't see the USB drive at boot up, then make sure you have the right bootable USB, reformat it and reflash the OS on it and check again.  
If it still doesn't work, then you can assure that it is a USB hardware issue.
FR, you can use SD card interface instead of a USB interface or you can replace the board with a new board.

nphamilton, 2017-06-05:
@keerockl I was thinking that would end up being the only solution, but I thought I'd give the forums a chance too. Thanks.

@vchagari I'm not using a USB drive. I am just trying to use a keyboard and it is not working. Additionally, I can't get to the BIOS because I cannot type any commands.

closed 2017-09-05 by keerockl