MinnowBoard-org/bugs-and-help #44: STEP File

Swisskbit, 2017-06-02:
Is there any STEP File available for the MinnowBoard.
We would like to get for a new development where
we are going to use that board.

Thanks for help.

Best regards,

Kurt

jnavarro7, 2017-06-08:
Hi, 
The step file for the Turbot_F200 used to be available online, but I can't find it anymore.  What is available is this: 
https://github.com/MinnowBoard-org/design-files/blob/master/minnowboard-turbot-dual-core/pcb-f200-revX205_Turbot-A/Turbot_F200_X205_VX/PCB/Layout/3D/Turbot_F200.xtda
Which from what I could find is a 3D file format for Mentor Graphics Xpedition. 
Mentor Graphics offered the exported .step file which was available in the previous wiki but I can't find it anymore. 
If you want I can upload the .step file somewhere and give you the link for download, just let me know. 

-Jose

MikeOnTheWork, 2017-06-08:
Hi José,

Can you send it to me too

Thanks

https://github.com/MinnowBoard-org/design-files/blob/master/minnowboard-turbot-dual-core/pcb-f200-revX205_Turbot-A/Turbot_F200_X205_VX/PCB/Layout/3D/Turbot_F200.xtda

https://github.com/MinnowBoard-org/bugs-and-help/issues/44#issuecomment-307161149

jnavarro7, 2017-06-08:
Hi, 
You can download the Turbot_F200.step file from here:
https://www.dropbox.com/s/qhasankrusgcrb8/Turbot_F200_Board_STEP.zip?dl=0

Regards
-Jose

Swisskbit, 2017-06-09:
Thanks for the great help.

Best regards,
Kurt

closed 2017-09-05 by Mark-van-der-Pol

roroserpi, 2017-10-04:
Hello,
Is it possible to have the Step file too, and the dxf top view if exist ?
Best Regards
 Romain

BrianOttaway, 2017-10-04:
FYI: 
.dxf file added to the design files repo for [Turbot B Dual-Core](https://github.com/MinnowBoard-org/design-files/tree/master/minnowboard-turbot-dual-core/pcb-F400_R302_Turbot-B) and [Quad-Core](https://github.com/MinnowBoard-org/design-files/tree/master/minnowboard-turbot-quad-core/F400_R302):. Len at ADI did a direct file export. 

We'll have to be sure to proactively provide .dxf and .step for future boards. 

-Brian