MinnowBoard-org/bugs-and-help #47: Display not detected in Minnowboard Turbot

mrajaraghavan, 2017-07-01:
Trying to connect a VGA Monitor with Minnow Board Turbot using Micro-HDMI to VGA Converter but display is not getting detected. I have installed Ubuntu 16.04 LTS in the board.

I have also observed that during booting if I connect a display with hdmi port using direct HDMI to micro-HDMI cable, the system is identifying the display make & resolution and working fine. After booting if I change the cable with micro-HDMI to VGA converter, the VGA monitor is also working fine (but in Settings->Display its not detecting any display even though i pressed Detect Display option). If I reboot the system with same setup, the display is not turning up also.

I have ascertain the said problem by changing micro-HDMI to VGA converter also and still the same above situation persist.

Kindly help me in this regard.

Mark-van-der-Pol, 2017-07-02:
There is a known issue on the MinnowBoard Max regarding driving the HDMI port with symptoms exactly as you describe, Will you confirm you are using a Turbot versus a MAX board, please? At first glance, they look the same with things like the heat sink maybe being different color.

Thanks, |\/|ark.

mrajaraghavan, 2017-07-03:
Thanks for your reply Mark and I confirm I'm using Minnowboard Turbot

BrianOttaway, 2017-09-05:
Hello @mrajaraghavan . Touching base on this one. Did you resolve the issue or move to an HDMI monitor? I'd like to know if the thread should stay open.

BrianOttaway, 2017-09-13:
@mrajaraghavan : I'm closing this issue. Haven't heard back from you. Feel free to reopen if needed.

closed 2017-09-13 by BrianOttaway