MinnowBoard-org/bugs-and-help #51: Minnowboard SPI Access

Donny3000, 2017-07-19:
I'm currently working on a project that needs to make use of the SPI interface of the Minnowboard (I'm using the quad core variant, if it matters), but I don't see any spi devices under _/dev_ (I'm running Ubuntu 14.04 by the way). I tried compiling the kernel module found here https://github.com/MinnowBoard-Projects/minnow-max-extras against my installed kernel, but didn't have any luck. The module would compile fine using the DKMS build system, but every time I would modprobe the driver I would get the following error:

> low-speed-spidev: module init
 low-speed-spidev: master=          (null)
 low-speed-spidev: Failed to register SPI device

Looking at the module code, it looks like I have to specify which SPI bus and CS line to use, but I don't know where I can find that information. I can continue going down this route of modify the kernel to get the SPI interface working, but I feel that I'm veering down a rabbit hole and there should be a simpler solution.

Is there another less involved method for gaining SPI access that I'm overlooking?

Mark-van-der-Pol, 2017-07-19:
After building the low-speed-spidev, did you install it into the kernel with sudo insmod low-speed-spi-dev.ko

The defaults should just work.
The only other thing to adjust is the default maximum speed - it is conservatively set at 1.0 MHz, and can be much higher - I am not sure what the max the silicon implements, but 16MHz or 25MHz seem within reason.

Then I run MRAA on top of that and the example source code floating around for SPI and MRAA worked as advertised.

So the steps you are taking are what should be taken, and possibly there is just a simple misstep we can identify and correct.

Let me know if you make progress with the information I provided.

Regards,
|\\/|ark .

Donny3000, 2017-07-31:
Mark,

Sorry for the extreme delay in responding to your comments, but I was issuing the sudo insmod low-speed-spi-dev.ko command that you mentioned. Funny thing is, when I went back and issued that command again after reading your comment, it all of a sudden worked. Then, to confirm consistency, I removed the module and reinserted it and still no error. So, I must have done something wrong or had a funky environment at the time. But, everything is working as expected following the directions provided. Thanks again for following up and assisting with my issue!

-Donald

closed 2017-07-31 by Donny3000