MinnowBoard-org/bugs-and-help #54: Help with EDK2

mikzaq, 2017-07-27:
This may be a little off-topic, but EDK2 is the firmware found for the minnowboard here:
https://firmware.intel.com/projects/minnowboard-max 
I figured maybe someone would be able to help.

I am trying to remove the ability to use the BIOS-like menu on my minnowboard, which is running EDK2 firmware:
https://github.com/tianocore/edk2-platforms/tree/minnowboard-max-udk2015
However I have been unable to locate anywhere that I can turn this capability off. Is there a way to do this?

Failing this, I would like to be able to set a BIOS password, and by going to BIOS > Device Manager > System Setup > Security Configuration, I see the option to "Setup Administrator Password", but setting this password does not seem to do anything, and I see no option to enable the usage of the password.

Mark-van-der-Pol, 2017-07-27:
Couple of things -
there is a known issue with the admin password in the current release - it is being resolved.

In any case, here are settings that you can use to achieve this, as follows:

Enter the Bios Menu. Choose Device Manager -> System Setup -> Boot
and change Fast Boot to <enable>
and change Silent Boot to <enable>

This makes the BIOS Menu inaccessible during boot time with keyboard inputs, no visual indication is given and (only at powerup) a brief Tianocore splash screen is shown before the previously selected boot loader is run. If there is no boot loader found it fails through into the UEFI Shell.

CAUTION: there is no easy way to get back into the BIOS Menu to alter the boot order, do a firmware update or turn this off.

Hope that helps.
|\\/|ark.

mikzaq, 2017-07-27:
I'm aware of the method of disabling the keyboard using Fast Boot, but I am
concerned that this can be bypassed by the following technique:

*Power button recovery*

The power button can be used to recover if you encounter Fast Boot problems.

   - From a powered off state, press and hold the power button for about
   two seconds, release it when you hear three short beeps.
   - The BIOS follows a normal boot path, as if Fast Boot is disabled.
   - If Fast Boot is enabled, the following message displays: Fast Boot
   failsafe trigger was detected. Would you like to restore Fast Boot on the
   next boot? (Y/N).

BrianRichardsonIntel, 2017-07-27:
The “BIOS Setup” in EDK II is known as the Setup Browser. It’s basically a UEFI application built into the firmware.

Source: MdeModulePkg/Universal/SetupBrowserDxe/

The fastest way to remove setup is to comment out its entry from the project’s FDF file. FDF defines the binary components built into the flash image. So comment out this line in PlatformPkg.fdf …

INF MdeModulePkg/Universal/SetupBrowserDxe/SetupBrowserDxe.inf

… and this line in PlatformPkgX64.dsc under the ‘Console’ section.

                 MdeModulePkg/Universal/SetupBrowserDxe/SetupBrowserDxe.inf

… then rebuild.

Removing setup means you may have to change default values for platform behavior, since you can’t go into a menu to change them. You may want to experiment in QEMU or NT32 emulator first.

Thanks … br
---
Brian Richardson, Senior Technical Marketing Engineer, Intel Software
brian.richardson@intel.com<mailto:brian.richardson@intel.com> -- @intel_brian (Twitter & WeChat)
https://software.intel.com/en-us/meet-the-developers/evangelists/team/brian-richardson

https://firmware.intel.com/projects/minnowboard-max

https://github.com/tianocore/edk2-platforms/tree/minnowboard-max-udk2015

mikzaq, 2017-07-27:
That worked! Thanks!

closed 2017-07-27 by Mark-van-der-Pol