MinnowBoard-org/bugs-and-help #55: How do I get OpenCL working with the Minnow Turbot GPU (assuming it's even possible)?

pkukol, 2017-08-05:
I've tried the graphics driver update tool as well as the big script that downloads / patches / builds the Linux kernel and a few other things, and no matter what I do I only ever see the CPU platform/device in OpenCL on my Turbot (no sign of the GPU, which is what I need). I'm using Ubuntu 16.04 though I'd be happy to switch to something else, if that's what it takes to get OpenCL with the GPU to work. I did have to fight the Ubuntu / libgles1-mesa issue, BTW - apparently xenial-updates has a version mismatch problem with libgles1-mesa vs. libglapi-mesa, and the only way I found to make things 'work' was to convince apt that I had the older version of libgles1-mesa installed (the hope being that the drivers only need libgles2-mesa anyway and the driver update tool's insistence on installing libgles1-mesa is a red herring). Any pointers? I don't care if the Turbot is officially supported by Intel as far as the OpenCL / GPU path or not, BTW - even if it works only 'unofficially', I'd be hopefully unblocked. TIA!

gvancuts, 2017-08-07:
Hi @pkukol , can you confirm what OpenCL package you installed (or rather, are trying to)? Are you using [Beignet](https://www.freedesktop.org/wiki/Software/Beignet/) or something else?

pkukol, 2017-08-07:
Thanks for replying. I’ve tried a number of things, but I never explicitly installed a particular OpenCL package per se – I’ve just been installing graphics drivers. With most GPU’s I’ve been working with, that’s how OpenCL gets installed (i.e. there is usually no explicit package to add OpenCL that one has to install, it’s part of the “normal” driver install) though there are certainly exceptions (like the Imagination PowerVR stuff, etc). I am not using beignet ATM, just the drivers from Intel. Specifically, I’ve run intel-graphics-update-tool (which, after the rigmarole with libgles1-mesa, finally runs without errors but still doesn’t provide GPU OpenCL) and the big Intel SRB5 driver install script as well (the one that patches and builds a 4.7.x kernel), as well as a few other things. I do see the Intel CPU platform / device in OpenCL so at least part of it is working fine, but I need to run OpenCL on the built-in GPU. Any suggestions?

 

Thanks

PeterK

uhofemei, 2017-08-07:
Hi Peter, 

The Yocto based Intel IoT RefKit distro provides reference profiles for various different industry verticals. Amongst other things, it provides a computer vision profile, which pulls in OpenCV, libRealSense and Beignet (OpenCL) for Intel integrated graphics. While you may not want to use Yocto to build your distribution, I suggest you refer to the meta data to identify the correct upstream source packages and versions. 

Please take a look at: https://github.com/intel/intel-iot-refkit/tree/d37ca62bccc6dda7b7c5a9a7da02c2a76c04837e/meta-refkit-computervision

and 

https://github.com/intel/intel-iot-refkit/tree/d37ca62bccc6dda7b7c5a9a7da02c2a76c04837e/meta-refkit-computervision/recipes-opencl

Thanks!
Ulf

gvancuts, 2017-08-07:
> Thanks for replying. I’ve tried a number of things, but I never explicitly installed a particular OpenCL package per se – I’ve just been installing graphics drivers. With most GPU’s I’ve been [...]

It's been a while since I looked at the Intel Graphics driver package but a quick glance tells me it does not include the OpenCL library. Can you try to install it from the Ubuntu repo (not sure of the package name but `apt search beignet` is your friend there). Or you can also try to install the latest and greatest from the Beignet website.

pkukol, 2017-08-11:
Thanks for the suggestions, everyone. Right now beignet doesn't work for me - any non-trivial OpenCL stuff I try running on it, a crash is the result (if I select one of the other OpenCL platforms / devices I have on the same box, no problem - everything works fine, same it does on every other box I've tried); but, it is true that by now I've uninstalled and reinstalled beignet and the Intel stuff so many times, who knows what harmful turds they might have left behind. I'll dig deeper, and also try grabbing the very latest and greatest beignet.

One related question that I still don't have a definitive answer for, though: what is the best OpenCL optimizer for Intel GPU's today? I've worked on compilers for 35 years so I know how hard it can be to tune these beasts ;). Do the drivers from the Intel web site (the ones that I'm still unable to get working with the GPU, only the CPU) use the same GPU optimizer that is in beignet, or do they have a non-open-source compiler in them? If beignet is the best there is today, it should be easy - I'll figure out why it's always crashing on the stuff I'm trying to run (worst case, I repave my box and start over with a fresh OS install) and I'll be off. But, the basic goal is to measure the best there is out there for OpenCL -> Intel GPU, even if it may not be the best long term choice (for whatever reason - technical, political, or otherwise).

TIA!

Peter