MinnowBoard-org/bugs-and-help #56: How to boot previously installed image on SD card?

hongyuli, 2017-08-10:
When we installed two copies of Ubuntu, the second copy made the first one not bootable. How can we get both running?

The process we did:
1. Follow the official tutorial install an Ubuntu 16.04 on the SD card.
2. After the installation(Ubuntu is running and bootable), power off the board and take off this SD card, plug in a second SD card.
3. Follow the official tutorial again install an Ubuntu 16.04 on the second the SD card.
4. But when we switch back to the first SD card, the system could not boot and only the second SD card is bootable.

It looks like the system installation overwrite something in the firmware, so that the previous system images you use is no longer discoverable. How can we get both/all the system images running?

Mark-van-der-Pol, 2017-08-10:
When you say in step 4 - the system could not boot, did you go into the boot menu and don't see the SD Card as a boot option?

If you go into the shell, does it list fs0: as a detected filesystem?

I typically have no issues switching SD Cards to boot into different OS', but it sometimes doesn't automatically boot, and I have to run the OS Bootloader manually. But that is now so automatic for me, I don't notice when I need to do that.

|\\/|ark

hongyuli, 2017-08-11:
The boot option menu of boot manager does not have the SD card, but the mapping table does show the fs0.

According to your hint, I googled on line and figured this out. In my case, when the UEFI shell shows up, use the following commands:
1. fs0: (enter this directory)
2. dir
3. cd EFI/ubuntu
4. grub.cfg (run this boot loader)

Thanks for your quick response and help. Let me close this issue.

closed 2017-08-11 by hongyuli