MinnowBoard-org/bugs-and-help #57: Experiencing Ubuntu 16.04 freezes frequently

hongyuli, 2017-08-10:
We follow the official tutorial to install the Ubuntu 16.04, but the system freezes frequently (around every 2 hours). There is no errors pops up through the Ubuntu's graphic UI, it just freezes randomly.
 
1. This might be the hardware issue. We may unfortunately receive the defective board. What kinds of test will be good for verifying the hardware errors?

2. Except Ubuntu 16.04, what other versions of Ubuntu or Linux distribution is stable on Minnowboard?

Mark-van-der-Pol, 2017-08-10:
Some details would help - 

- Are you running the x64 or i32 version of Ubuntu?
- What board version and vendor are you using? MAX, Turbot or Turbot Quad? or other?
- What specific distribution are you using? 16.04 has had updates, and the most recent is 16.04.3
- Have you updated the running system using the built-in update utility?
- What is the system doing for the 2 hours before it freezes?
- Is it connected to the network or internet during this time?
- What lures and accessories are connected?
- What is the power supply rating?

One simple test is to connect to YouTube, and watch a video, and set up YouTube to automatically keep showing videos. This keeps the processor and graphics cores humming along and if it is a board stress or thermal issue, this would have it show up quickly.

Other linux version we know work are ubuntu 16.10 and 17.04, fedora, clear linux, CentOS, yocto, and others tested outside of our team.

Thanks, |\\/|ark.

hongyuli, 2017-08-11:
Running x64 Ubuntu, exactly following the this official tutorial https://minnowboard.org/tutorials/installing-ubuntu-16.04-on-minnowboardmax
Turbot QUAD Core Board
SD-Card, SanDisk 64GB, Class 10 (we have multiple sd-cards, all have the same issue)
We firstly try the 16.04.1 as suggested the previously mentioned official tutorial. We also tried the apt-get upgrade to 16.04.3. Both have the same issue.
Basically we installed bro, and tried to use it to monitor some traffic. The log files were not large.
Internet is connected.
The official accessories and power supply(5W 4A P/S) from https://store.netgate.com/Turbot4.aspx

We ran a memory test software https://www.memtest86.com/, and it shows 55 errors were detected. Could this confirm the hardware issue?

Mark-van-der-Pol, 2017-08-11:
A couple of things - the tutorial should have been updated to refer to at least 16.04.02, as there are issues with the 16.04.01 and earlier releases on the Turbot. And it is possible that an upgrade to 16.04.3 is subtly different than installing 16.04.3. I will take care of updating the website, and thank you for bringing that to my attention.

Memory tests should have 0 errors, so if it is finding any that is a concern. There may also be a memory test option in the grub boot loader before Ubuntu starts - run that as well to confirm memory errors.

If these memory errors persist, then it could account for the crashes. 
 
Thanks, |\\/|ark.

BrianOttaway, 2017-09-05:
See issue #60 for investigation and root cause on false positive memtest86 results.

BrianOttaway, 2017-09-13:
@hongyuli : are you still having trouble?...should we close the issue?

hongyuli, 2017-09-14:
We have returned the board, thanks for following up.

closed 2017-09-14 by hongyuli