MinnowBoard-org/bugs-and-help #58: Reflashing MinnowBoard MAX with DediProg SF600 and DediProg software v.6.0.5.18

mdlemay, 2017-08-29:
My MinnowBoard MAX recently stopped booting.  It no longer displays any output.  So, I decided to try reflashing it.  I attached the ISP cable via the DediProg SF600-to-SF100 universal adapter.  The interface for the new version of the DediProg software is significantly different from what is shown at https://minnowboard.org/tutorials/update-firmware-using-spi-tool, but I did my best to follow the instructions.  I attempted to program the flash both in interactive mode entirely through the software interface rather than using batch mode and also in batch mode, but neither worked.  The interactive approach seems simpler, so I'll describe what I tried along that route:
1. I pressed the "Select" toolbar button and selected the W25Q64 chip.
2. I pressed the "File" toolbar button and selected the `MNW2MAX1.X64.0096.R01.1706301843.bin` file.
3. I pressed the "Prog" toolbar button.

The following error messages appeared:
[image]

What might I be doing wrong? Thank you.

Mark-van-der-Pol, 2017-08-30:
I don't think you are doing anything wrong.

At first blush it looks like a catastrophic failure on the board, or at least in the flash part.

The programming interface and DediProg are designed to program the part totally isolated from the board - that is, power and signals are all provided by the programming interface. Nothing is needed on the board as long as the signals get to the part and have integrity.

There is a chance that the programming interface itself experienced a failure- I don't know if there is a self-diagnostic that can be run in the DediProg - but barring that, the part not being recognized does not bode well. 

|\\/|ark.

BrianOttaway, 2017-09-13:
@mdlemay : did you figure out the problem?

mdlemay, 2017-10-06:
Sorry for forgetting to respond to this.  Unfortunately, I was unable to determine why reflashing failed.

BrianOttaway, 2017-10-06:
A sad conclusion. If you have any more to share, just reopen the thread.

closed 2017-10-06 by Mark-van-der-Pol