MinnowBoard-org/bugs-and-help #59: XEN doesn't boot on Minnowboard

gogo9th, 2017-09-03:
Hi, 

I installed XEN 4.8 on Minnowboard (sudo apt install xen-hypervisor-amd64), but DOM0 kernel images based on XEN hypervisor don't boot up at all. Does anyone have the same issue? Is Minnowboard incopatible with Minnowboard? Please share with me any clue... Thanks!

gogo9th, 2017-09-06:
When I boot Linux 4.4.0-92-generic XEN 4.6-amd64 hypervisor, I get the warning:

WARNING: no console will be available to OS
Loading Linux 4.4.0-92-generic ...
Loading initial ramdisk ...

Then, after 10 seconds, the whole texts above get shifted to some random location on the screen and I get two horizontal rainbow lines on the screen.

gogo9th, 2017-09-07:
Here are more updates. Apparently, XEN 4.9 started servicing EFI boot with GRUB2: 
https://wiki.xenproject.org/wiki/Xen_Project_4.9_Feature_List
 
But if I install xen with apt install, its version is 4.8. So I downloaded XEN source code from here: 
https://xenproject.org/downloads/xen-archives/xen-project-49-series/xen-project-490.html

and installed XEN following this tutorial:
https://mhsamsal.wordpress.com/2013/12/05/configuring-virtual-tpm-vtpm-for-xen-4-3-guest-virtual-machines/

However, when I run XEN 4.9, I still get exactly the same message as before:

WARNING: no console will be available to OS
Loading Linux 4.4.0-92-generic ...
Loading initial ramdisk ...

Did anybody succeed in running XEN 4.9 on Minnowboard? Could someone please help me out?

gvancuts, 2017-09-07:
I've never tried myself but when googling for possible hints I saw someone who mentionned he was struggling with the initramfs and ended up removing that from the OS he was trying to boot.

gogo9th, 2017-09-07:
Thanks for the feedback and I finally figured out a solution based on this: https://wiki.xenproject.org/wiki/Xen_EFI

However, this tutorial is incomplete in some way, so here's my detailed walkthrough:
- vTPM Tutorial: https://mhsamsal.wordpress.com/2013/12/05/configuring-virtual-tpm-vtpm-for-xen-4-3-guest-virtual-machines/

[XEN Source Code Installation]
- Xen 4.9 Source Code: https://www.xenproject.org/downloads/xen-archives/xen-project-49-series/xen-project-490/351-xen-490

sudo apt-get install bcc bin86 gawk bridge-utils iproute libcurl3 libcurl4-openssl-dev bzip2 module-init-tools transfig tgif texinfo pciutils-dev mercurial build-essential make gcc libc6-dev zlib1g-dev python python-dev python-twisted libncurses5-dev patch libvncserver-dev libsdl-dev libjpeg62-dev iasl libbz2-dev e2fslibs-dev git-core uuid-dev ocaml libx11-dev bison flex xz-utils ocaml-findlib gcc-multilib checkpolicy libglib2.0-dev g++-multilib liblzma-dev libpixman-1-dev libaio-dev libyajl-dev pandoc markdown

# sudo apt-get install texlive-latex-base texlive-latex-recommended texlive-fonts-extra texlive-fonts-recommended

--- "Config.mk" ---
...
XSM_ENABLE ?=y
FLASK_ENABLE ?=y
...
------------------------

--- "tools/test/xenstore/xs-test.c" ----
...
int a;
...
a = asprint(...)
...
-------------------------------------------

sudo make xen
sudo ./configure
cd tools
sudo ./configure
cd ..
sudo make tools
sudo make stubdom
cd stubdom
sudo make
cd ..
sudo make install-xen
sudo make install-tools PYTHON_PREFIX_ARG=
sudo make install-stubdom
sudo make -C tools/flask/policy
sudo ldconfig					# To update XEN’s .so libraries
sudo cp dist/install/usr/lib64/efi/xen* /boot 	# !!! Otherwise, UEFI can’t start xen
sudo update-grub
------------- Modify: “/etc/xen/xend-config.sxp” -----------------
…
(xend-unix-server yes)
…
----------------------------------------------------------------------------

[Botting xen.efi directly with UEFI]
- Launching XEN.efi with UEFI directly: https://wiki.xenproject.org/wiki/Xen_EFI

sudo mkdir /boot/efi/EFI/xen
sudo cp /boot/vmlinuz* /boot/efi/EFI/xen
sudo cp /boot/initr* /boot/efi/EFI/xen/
sudo cp /boot/xen-4.9.0.efi /boot/efi/EFI/xen

------- Create: "/boot/efi/EFI/xen/xen-4.9.0.cfg" --------------
[ubuntu]
options=console=vga,com1 com1=115200,8n1 iommu=verbose ucode=scan flask=disabled conring_size=2097152  loglvl=all
kernel=vmlinuz-4.8.0-41-generic root=UUID=3f1e35fb-9907-48d1-b621-42369d5ad88f ro quiet vt.handoff=7 console=hvc0
ramdisk=initrd.img-4.8.0-41-generic
--------------------------------------------------------------------------
# vmlinuz, initrd version should be your own kernel version. UUID should be your kernel's root partition's UUID. Check these values from "cat /proc/cmdline". 

# This specifies the location of xen.efi. In this case, the EFI partition is in /dev/sda1, first partition, and xen.efi is under /boot/efi/EFI/xen directory. 
efibootmgr -w -L Xen -d /dev/sda1 -p 1 -l "\EFI\Xen\xen.efi" -c  

[Start XEN]
sudo reboot

sudo /etc/init.d/xencommons start    
sudo /etc/init.d/xendomains start    
sudo /etc/init.d/xen-watchdog start    
sudo /etc/init.d/xendriverdomain start
sudo xl list 					# Check if DOM0 is running

gogo9th, 2017-09-07:
Thanks and I finally found the solution.

- vTPM Tutorial: https://mhsamsal.wordpress.com/2013/12/05/configuring-virtual-tpm-vtpm-for-xen-4-3-guest-virtual-machines/

[XEN Source Code Installation]
- Xen 4.9 Source Code: https://www.xenproject.org/downloads/xen-archives/xen-project-49-series/xen-project-490/351-xen-490

sudo apt-get install bcc bin86 gawk bridge-utils iproute libcurl3 libcurl4-openssl-dev bzip2 module-init-tools transfig tgif texinfo pciutils-dev mercurial build-essential make gcc libc6-dev zlib1g-dev python python-dev python-twisted libncurses5-dev patch libvncserver-dev libsdl-dev libjpeg62-dev iasl libbz2-dev e2fslibs-dev git-core uuid-dev ocaml libx11-dev bison flex xz-utils ocaml-findlib gcc-multilib checkpolicy libglib2.0-dev g++-multilib liblzma-dev libpixman-1-dev libaio-dev libyajl-dev pandoc markdown

# sudo apt-get install texlive-latex-base texlive-latex-recommended texlive-fonts-extra texlive-fonts-recommended

--- "Config.mk" ---
...
XSM_ENABLE ?=y
FLASK_ENABLE ?=y
...
------------------------

--- "tools/test/xenstore/xs-test.c" ----
...
int a;
...
a = asprint(...)
...
-------------------------------------------

sudo make xen
sudo ./configure
cd tools
sudo ./configure
cd ..
sudo make tools
sudo make stubdom
cd stubdom
sudo make
cd ..
sudo make install-xen
sudo make install-tools PYTHON_PREFIX_ARG=
sudo make install-stubdom
sudo make -C tools/flask/policy
sudo ldconfig					# To update XEN’s .so libraries
sudo cp dist/install/usr/lib64/efi/xen* /boot 	# !!! Otherwise, UEFI can’t start xen
sudo update-grub
------------- Modify: “/etc/xen/xend-config.sxp” -----------------
…
(xend-unix-server yes)
…
----------------------------------------------------------------------------

[Botting xen.efi directly with UEFI]
- Launching XEN.efi with UEFI directly: https://wiki.xenproject.org/wiki/Xen_EFI

sudo mkdir /boot/efi/EFI/xen
sudo cp /boot/vmlinuz* /boot/efi/EFI/xen
sudo cp /boot/initr* /boot/efi/EFI/xen/
sudo cp /boot/xen-4.9.0.efi /boot/efi/EFI/xen

------- Create: "/boot/efi/EFI/xen/xen-4.9.0.cfg" --------------
[ubuntu]
options=console=vga,com1 com1=115200,8n1 iommu=verbose ucode=scan flask=disabled conring_size=2097152  loglvl=all
kernel=vmlinuz-4.8.0-41-generic root=UUID=3f1e35fb-9907-48d1-b621-42369d5ad88f ro quiet vt.handoff=7 console=hvc0
ramdisk=initrd.img-4.8.0-41-generic
--------------------------------------------------------------------------
# vmlinuz, initrd version should be your own kernel version. UUID should be your kernel's root partition's UUID. Check these values from "cat /proc/cmdline". 

# This specifies the location of xen.efi. In this case, the EFI partition is in /dev/sda1, first partition, and xen.efi is under /boot/efi/EFI/xen directory. 
efibootmgr -w -L Xen -d /dev/sda1 -p 1 -l "\EFI\Xen\xen.efi" -c  

[Start XEN]
sudo reboot

sudo /etc/init.d/xencommons start    
sudo /etc/init.d/xendomains start    
sudo /etc/init.d/xen-watchdog start    
sudo /etc/init.d/xendriverdomain start
sudo xl list 					# Check if DOM0 is running

gvancuts, 2017-09-11:
Thanks for sharing your successful steps!!

BrianOttaway, 2017-09-13:
Thx all. Closing this one up.

closed 2017-09-13 by BrianOttaway