MinnowBoard-org/bugs-and-help #60: Memtest86 Fails on MinnowBoard Turbot

BrianOttaway, 2017-09-05:
We heard several reports from community members that running Passmark's popular [Memtest86 ](https://www.memtest86.com/)tool on a MinnowBoard Turbot produces failures that typically suggest a hardware memory issue (faulty memory). This thread will follow the issue investigation and response.

keerockl, 2017-09-06:
Memtest86 tool is calling one specific function with rare but legitimate NULL value(MP protocol StartupThisAP with a NULL *Finished parameters) which triggers the memory test error.
This is not hardware/memory issue and takes effect only in BIOS boot time, which means there is no impact on operating systems running on the board.
Next version of FW image will include this fix in next release. 

Please refer to below link for this topic in passmark forum.
https://www.passmark.com/forum/memtest86/40329-minnowboard-turbot-multi-cpu-failures

BrianOttaway, 2017-09-13:
If you ran a Memtest86 test and encountered this issue please do toss a comment in here. I'm curious how many developers actually saw it.

ftab, 2017-09-15:
I experienced lockups during OS run time - on two different flavors of Linux - which happened at random, sometimes within minutes of booting and sometimes not until hours later. I ran Memtest86 which seemed to reveal a cause in that the first byte of memory was frequently incorrect in some tests when running in multi core mode. This made sense to me at the time because Linux will use multiple cores, so I thought it might be exposing a flaw in firmware that handles multi processing. You’re saying that Memtest86 uses a different mode that the OS does not use? Does that mean there is a different issue causing lockups in the OS and that I just happened upon an unrelated issue seeing those errors in Memtest86?

keerockl, 2017-09-15:
@ftab Yes that would be two different issues. first we will update BIOS to fix memtest86 issue and then further check if lockup issue still persist.

Mark-van-der-Pol, 2017-09-15:
@ftab - we believe there is no correlation between the memtest86 faux error report, and your lock up issue. Our root cause analysis detailed above shows it is an API mis-call.

So your issue is really separate and should be transposed to a new thread - feel free to do that, and please report the Linuxes you experience this on, and what they are running from - USB flash, USB HHD or USB SSD , SATA HDD or SATA SSD, or SD Card. And the capacities of those media, speed rating and manufacturer.

Pro-actively asking, as waiting for the new F/w, and finding you still have the issue, as I expect we'll find, looses this time we could be delving into a separate issue.

Thanks, |\\/|ark.

keerockl, 2017-10-04:
FW 0.97 is out and available public in below link. This has fix for memtest86 fail issue during boot up time.
https://firmware.intel.com/projects/minnowboard-max

closed 2017-10-04 by keerockl