MinnowBoard-org/bugs-and-help #61: fTPM implementation on Minnowboard Turbot seems to be stuck at Rev 0.93 which is very outdated

mms-, 2017-09-06:
MinnowBoard Max/Turbot fTPM revision is 0.93 which is outdated and does not support all necessary algs for actual TPM 2.0 spec. Any idea when this will be updated? Intel has updated the firmware, mentions fTPM but implementation has been stuck at Rev 0.93 for a long time now. This makes the fTPM unusable.

https://firmware.intel.com/projects/minnowboard-max

zwei4, 2017-09-08:
The latest UEFI BIOS is V0.95. What specific TPM 2.0 features do you think are missing?

naikrovek, 2017-09-08:
0.96 is the latest, as of June.

Windows 10 IoT doesn't see enough capabilities in the TPM to show useful details or enable the Azure integration that a fully supported TPM would provide.

mms-, 2017-09-08:
Yes the latest firmware is 0.96 which I have, part of that firmware the fTPM is at revision 0.93 which is old, and supposedly newer builds are available out there, would be good to see Intel update this part of their builds so full TPM 2 capabilities are available. I think specifically some algorithms are missing which are required by full TPM2.0 and therefore Windows IoT.

BrianOttaway, 2017-09-13:
@zwei4 : any thoughts on how big the gap is on the fTPM to support WinIOT? I know somebody on Microsoft IOT side that could possibly help out.

mms-, 2017-09-13:
Looks like it's missing SHA256-HMAC support, more info here : https://social.msdn.microsoft.com/Forums/en-US/5e04e461-768e-4e79-88ec-56e3ceb4c68f/tpm-how-to-enable?forum=WindowsIoT 

An updated fTPM implementation likely already exists but seems it won't be released for this now older SoC.

naikrovek, 2017-09-13:
You can always put an external TPM on the GPIO header and disable the fTPM on the MBM.

I designed a TPM board for the Raspberry Pi and it could easily be adapted for the MBM if the fTPM is truly abandoned on that board.

mms-, 2017-09-13:
I see, where do you buy the TPM?

naikrovek, 2017-09-14:
The TPM devices are sold as bare chips and require soldering onto a circuit board that connects the correct pins on the chip to the correct pins on the GPIO header of the device in question.

I don't sell any completed boards. There isn't a huge market for Raspberry Pi TPMs and as I said, I haven't created a MBM version, yet.

mms-, 2019-01-16:
Haven't looked into this in a while, perhaps it is updated now? Who ever is in charge of putting together the firmware for this device simply needs to update the fTPM component for the next firmware release. Not sure who does that. Is the firmware on github?

JacquesCC, 2019-01-17:
I have had no luck in finding any information on the fTPM firmware. Is there anyone else that can confirm for us and also possibly point me in the right direction?