MinnowBoard-org/bugs-and-help #62: Unable to update firmware with error 'SPI Protocol is now found.'

mms-, 2017-09-08:
After flashing the MinnowBoard Turbot with Intel Firmware Engine 3.0 generated firmware, the board boots however I'm unable to flash new firmware with the error 'Spi Protocol is not found' given by the standard .efi and firmware provided here : https://firmware.intel.com/projects/minnowboard-max

Somehow the firmware from IFE does not include SPI capability? Anyway to add this from within UEFI Shell? Or is the only way to at this point is a Dediprog sf100?

Also the IFE firmware does not come with a UEFI GUI? I only have the option of the shell.

Thanks for any help.

BrianRichardsonIntel, 2017-09-08:
If you flash an image generated with Intel Firmware Engine onto the platform, you cannot use the software utilities provided with the open source project to flash a new image. Differences in SPI ROM headers generated by the firmware build tools mean the binary images are not interchangeable.  An SPI programmer like the DediProg SF100 or TinCanTools SPI Hook is required to switch between the EDK II images and Intel Firmware Engine images.

Both Intel Firmware Engine and the EDK II open source project support the SPI ROM, but their use different software-based update methods. This is why the UEIF Shell Firmware Update utility for EDK II MinnowBoard Max/Turbot won’t work with an Intel Firmware Engine image (or vice versa).

The Intel Firmware Engine or EDK II project do not provide a GUI pre-OS menu for UEFI. Both have the option for the UEFI Shell or a text-based firmware setup menu. Several commercial vendors offer UEFI GUI menus, but there is not one available yet in open source. Hopefully someone will be motivated to contribute this to the TianoCore project.

Thanks … br
---
Brian Richardson, Senior Technical Marketing Engineer, Intel Software
brian.richardson@intel.com<mailto:brian.richardson@intel.com> -- @intel_brian (Twitter & WeChat)
https://software.intel.com/en-us/meet-the-developers/evangelists/team/brian-richardson

mms-, 2017-09-08:
Thanks for the detailed info. I didn't realize the two were so different. I don't need to switch back to the EDK II based firmware, just need to be able to re-flash with correct settings.

Now that I'm on the IFE based firmware, how do I re-flash again with another IFE binary?

Also I think after flashing with the IFE based image the firmware is 64bit, I need to boot a 32bit OS, how to specify the platform to build Intel Firmware Engine? 

Right, by GUI I meant the text based old school blue BIOS setup UI, I can't get into it anymore, 'exit' in the shell just launches the shell, where as in the EDK II firmware it would go into the UI.. Hitting f2, delete, etc on boot does nothing. I did have 'fast boot' enabled in the IFE project, also had the 'enter firmware user interface support' option checked.

On the question of going back to EDK II firmware. If I was able to flash an Intel Firmware Engine image from within the EDK II shell, there should be an executable to allow me to flash an EDK II image from within Intel Firmware Engine shell? If it's just a binary image that is written, then all I need is a Intel Firmware Engine based executable to flash whatever image?

Thanks again.

CHMcFarlandIntel, 2017-09-13:
mms,

Let me try to clarify and help here.  Your original assertion/question is correct.  Intel Firmware Engine does not have support for a GUI based setup.  Setup type changes require making the changes in the Intel Firmware Engine tool and re-flashing.

Signed Capsule Update is the official, secure way to re-flash and Intel Firmware Engine firmware image. The update covers DXE image on the flash. PEI will not be updated due to recovery concerns.

Dediprog is another option if you have physical access to the board.  And allows you program the entire flash part as well as switch between different binaries.

To switch to 32 Bit support, simply make the change in the tool GUI, rebuild and flash.  The switch is as shown in the below picture.

[image]

Thanks,
Chris

mms-, 2017-09-13:
Thanks for that, that dialog answers all my questions, but is not as easy to find as it should be, UI could be better.

I started a new project, to make sure I don't have any unwanted changes, checked off 'enable firmware update file generation', build and flashed from within UEFI shell using : CapsuleApp.efi FVMAINCAPSULE.Cap

it ran, but before any output could be seen the board reset, now it is bricked with no video output, power LED is and button is responsive.

I had the .FD file in the folder as well, FVMAINCAPSULE.Cap itself was only 1.3MB.

BrianOttaway, 2017-10-04:
@mms- Did you resolve the issue? Continuing problem? From you last comment it sounded like you needed to flash with dediprog, or similar.

mms-, 2017-10-04:
Yes it is bricked, don't have a dediprog to do it. Just wondering why flashing with the IFE generated firmware bricked the device.

naikrovek, 2017-10-16:
You can probably flash the SPI flash chip on the MBM with a Raspberry Pi, if you have one of those.  the 'flashrom' package can almost certainly get you going again.  

As to why it failed, I think it's because your .fd file was too small.  All the ones I've generated have been the same size -- 8MB.