MinnowBoard-org/bugs-and-help #63: Unable to boot from USB or microSD

hotplot, 2017-09-09:
Hi,

I have a MinnowBoard Turbot Dual-E with latest firmware (MNW2MAX1.X64.0096.R01.1706301843) and cannot get it to boot from any device

It is powered from a 5V/3A supply

When I insert a bootable USB3 stick in the bottom USB port and a wireless USB keyboard dongle in the top port, the UEFI shell displays "no mapping" and there is no USB or Misc boot device option in the UEFI boot devices menu. Typing is also very slow - almost all key presses are missed, but I can eventually type `exit` to get to the menu at which point the keyboard seems to become responsive again.

When I swap the USB stick and dongle around (so the USB stick is in the top port), I can type without issue and the UEFI shell shows a mapping for BLK0-2. Executing `ls` in any of those blocks results in `ls: File Not Found: BLK0:\` and trying to boot from USB in the UEFI menu results in a very brief `Boot failed. Misc USB device` message before returning to the menu.

I can boot from the same USB stick using my MBP, so I am certain that it is bootable and not corrupted.

The USB ports must be working to some extent since I used a USB stick to upgrade the firmware. Behaviour was the same with the April firmware - unable to boot from any device.

Mark-van-der-Pol, 2017-09-12:
That is very odd behaviour, indeed. 

Telling you I don't see this is not going to help you, I imagine. And I will tell you that I have not seen anything like this - the USB 3 port works just like the USB 2 port, except it supports USB 3 devices and hubs...

Just to start eliminating obvious causes - the USB-3 stick, what is the pedigree? Could it be a fake with only 2,4, or 8 GByte masquerading as a 32, 64 or 128GB+ device?

What format is the stick using - UEFI requires GPT partitions, whereas MBP can deal with MBR or GPT or hybrid formatted boot media. 

|\\/|ark.

hotplot, 2017-09-12:
Hi Mark

Thanks a lot for your response. I have done some more testing and successfully booted my MinnowBoard, although there are still a couple of surprising issues that I don't understand:

- I have two Apacer USB sticks and one Lexar uSD card
- All three devices can have ISOs burnt using Etcher and pass verification
- All three devices can be booted from using my MBP
- The newest Apacer USB3 stick will not boot on the MinnowBoard (error as described above), but the older Apacer and the Lexar uSD both boot Ubuntu successfully

I am also unable to boot RancherOS, IPFire or OPNsense although those may well be failing due to some misunderstanding on my part. Ubuntu and Windows 10 IoT both boot successfully.

Anyway, thanks for your reply and feel free to close unless you have anything further to add

Mark-van-der-Pol, 2017-09-12:
Wow, that is most interesting. Can you give me the exact make, model and size of the Apacer USB3 that doesn't work? Best the exact order info so I can order one to investigate.

And I am relieved to hear that you are successful in booting Ubuntu and Windows. 

I would also love it f you would report on your findings getting these other OS's up and running - it seems that the dual Ethernet MinnowBoaards are hitting a spot and I am inclined to write a tutorial for one or more of these. You can either put it on the github here, or PM me.

Thanks, |\\/|.

closed 2017-09-12 by Mark-van-der-Pol

hotplot, 2017-09-20:
The markings on the troublesome 32GB USB stick are:

Apacer USB 3.1
AP32GAH356
961717507291

And this is the product page: http://consumer.apacer.com/eng/content.php?sn=951

Hope that helps a bit!

hotplot, 2017-10-02:
Hi Mark

I downloaded the RancherOS ISO from https://github.com/rancher/os/releases/tag/v1.1.0 and IPFire from http://downloads.ipfire.org/releases/ipfire-2.x/2.19-core113/ipfire-2.19.x86_64-full-core113.iso

Neither support UEFI and therefore fail to boot

I have not had time to experiment deeply at this point but I believe both could be booted by either finding a copy of the Winzent Minnowboard BIOS, compiling Coreboot or booting with iPXE. Hopefully I will have an opportunity to try these options in the next few weeks.