MinnowBoard-org/bugs-and-help #68: Turbot hard drive storage specs

MinimalistAudiophile, 2017-09-29:
I'm planning to attempt a Turbot dual-core as a music server, and would like to know if there are any restrictions on the size of a SATA hard drive it will recognize.  

Thanks.

BrianOttaway, 2017-10-04:
Turbot has 1x SATA2.0. As long as you have a somewhat recent 64-bit OS (e.g., not Windows 2k) you should be able to go to Petabyte space. :) Please do reply to thread with what you end up with...maybe a picture and brief setup tutorial? Others may do the same thing. 

NOTE: do be aware that the Turbot requires separate SATA power supply.

Mark-van-der-Pol, 2018-01-22:
Closing this - and updates regarding experiences with specific drives can always be added as they are tried.
|\\/|ark.

closed 2018-01-22 by Mark-van-der-Pol