MinnowBoard-org/bugs-and-help #69: Mechanical dimensions for Turbot Dual-Ethernet

BrianOttaway, 2017-10-04:
As submitted in direct MB.org email. 

> I have a question regarding the board layout of one of your products. A colleague of mine ordered a board and I need to design a plate that it will mount to. I took some rough dimensions just to verify it will fit where we want it but I will need precise information about the 6 mounting holes on this unit to ensure proper fit between this board and my plate. I have a serial number (S/N D901xx-xxxx) and also have attached a picture. Could you please pass this request on to the appropriate person?

BrianOttaway, 2017-10-04:
Turbot Dual-Ethernet F300 gerbers are located here: https://github.com/MinnowBoard-org/design-files/tree/master/minnowboard-turbot-dual-E-Dual-and-Quadcore/F300_R201 You can view there with a gerber viewer, or output to autocad .dxf. @MinnowBoardHW might get to it before I do.

MinnowBoardHW, 2017-10-04:
I would recommend gerbv, which is a great open-source gerber viewer: http://gerbv.geda-project.org/.  Let us know if you need anything else.

BrianOttaway, 2017-10-04:
Perfecto. Thx @lsoika . Closing this thread.

closed 2017-10-04 by BrianOttaway