MinnowBoard-org/bugs-and-help #72: MinnowBoard enable debug UART

krishna763, 2017-10-13:
Hi,
I am using minnowboard turbot(2GB Quad Core Intel Atom E3845 series SoC). I have followed this link to setup my serial console: [https://minnowboard.org/tutorials/ftdi-serial-connection](url). I am using Ubuntu 14.04 as my host machine. I am unable to see any logs on my host. I am not getting even bootloader logs.

How do I make my console up? Thanks for the help.

gvancuts, 2017-10-13:
What terminal emulator are you using? The tutorial indicates you should use 115200 8N1 with HW flow control turned off. I personally use `minicom` a lot and one thing that bit me time and again is that SW flow control *also* needs to be turned off else you don't see anything on the screen.

krishna763, 2017-10-13:
I am also using mincom with HW and SW flow both turned off.

I also tried with a C program which will open ttyS0 and write to it continuously. Still did not get anything.

gvancuts, 2017-10-13:
The serial port you need to use is most likely `/dev/ttyUSB0` (and not ttyS0). Can you check in dmesg to make sure you're using the right one? `dmesg | grep FTDI`

krishna763, 2017-10-13:
I was mentioning about ttyS0 commands from my minnowboard so that I will receive them on ttyUSB0 on my host system.

closed 2017-10-13 by krishna763