MinnowBoard-org/bugs-and-help #73: Heatsink/Fan not in BOM documents

MikuNeko, 2017-10-14:
I was unable to find the heatsink used in the dualcore and the heatsink+fan used in the quadcore boards reflected in the BOM documents:

Turbot_DUAL-E_Dual_Core_F300_R201-BOMRev201.xlsx
Turbot_DUAL-E_Quad_Core_F300_R201-BOMRev201.xlsx

May I know what heatsink/heatsink + fan are used for the boards?

Mark-van-der-Pol, 2017-10-16:
The heatsink and fansink on these boards are custom designs and built by Minimetal.

I am working on obtaining permission to release the design files for them, so you can derive your own design, or find generic parts that fit the profile.

And - coming up with robust under environmental extremes, long term stable thermal management for these SoCs is a complex science and Intel has extensive specifications to guide such designs. But for prototypes and simple one-off projects, you can likely get away with seat-of-the pants over-specifying the cooling.

|\\/|ark.

MikuNeko, 2017-10-16:
Hi Mark, thanks for the comment. Are you able to provide a part number and/or datasheets least?
I am looking to see if I can make a quadcore board go fanless.

Judging from measurements made using high-resolution board pictures, the diagonal distance between heatsink mounting holes is 59mm, and the Enzotech CNB-S1 copper heatsink appear to fit the profile.

Further testing would be needed to see if the CNB-S1 would be able to cope with the ~50% additional TDP of the quadcore CPU over the dualcore. (10W vs 7W)

Ideally I'd like to test and measure the following:
1. Dual-core, stock
2. Dual-core, CNB-S1
3. Quad-core, stock
4. Quad-core, CNB-S1

No point going fanless if it results in decreased performance/lifespan due to throttling/thermal extremes...

Any chance those Intel SoC thermal design guidelines are available to plebeian masses fielding small projects - such as yours truly?

Mark-van-der-Pol, 2017-10-23:
I just pushed the design sketches for the heat-sinks on the Dual-Core and Quad-Core Turbots. These same heat sinks are also used on the Dual-Ethernet derivative designs. I know these are not part numbers, and are in fact custom designs by ADI and built by Minimetal for Netgate.

https://github.com/MinnowBoard-org/design-files/blob/master/minnowboard-turbot-dual-core/pcb-F400_R302_Turbot-B/MM-32900-0020-20160908.pdf

https://github.com/MinnowBoard-org/design-files/blob/master/minnowboard-turbot-quad-core/F400_R302/MM-32900-0021-A1-20161019-updated.pdf

|\\/|ark.

closed 2017-10-23 by Mark-van-der-Pol