MinnowBoard-org/bugs-and-help #75: Quad core Turbot and I2S Audio

tranfa, 2017-10-28:
Good afternoon,

I recently purchased a quad core Turbot, in order to be able to use its I2S Audio capabilities.

Which OS should I use in order to fully exploit this characteristic?

Did anyone develop any Windows - Linux driver for this purpose?

Thank you very much!

Mark-van-der-Pol, 2017-11-01:
To the best of my knowledge, Intel sourced Linux kernels have support for Bay Trail I2S serial ports, and support 3 separate ports which would have to be probed to ascertain which are active. One of the I2S driver modules would have to be loaded, and then the audio application can use the I2S port to communicate with your I2S codec.

I have no further information at the moment, and I would love to have a tutorial on our new, updated web-site to cover this use case. Are you able to share with us what your application is and what you learn so I can leverage it into something useful for the entire community?

Regards, |\\/|ark.

tranfa, 2017-12-07:
Hi Marc,

I just answered to Chris along the same lines.

Just received my Turbot right now - which explains my delay in answering.

My purpose is to connect the Turbot to an AudioGD I2S out mini board, and therefore to a Dac which allows I2S in.

I’d like to use this as an Ethernet audio streamer, possibly arriving to dsd128 frequencies.

I’ll try with Ubuntu as per tutorial, and then with an AudioLinux which I know supports both I2S and my Dac.

Will keep you posted, and thank you in the meantime! 

Pietro

Inviato da iPhone

MinimalistAudiophile, 2017-12-23:
Hi Pietro,

That sounds very similar to what I plan on trying ... I've got the AudioGD I2S module, but not the minnowboard yet.  I plan to attempt a music server with I2S output using Audiophile Linux 4, which supports native DSD256.

Look forward to hearing how you make out.

Thom.

rodrigojilar, 2018-04-22:
Hi! 
I hace just received my board. And I'm facing the same doubt. I Installed Ubuntu 16.04 LTS and I would like to use I2S with PulseAudio. Any suggestions?
Thanks.

MinimalistAudiophile, 2018-04-22:
Hi there - You know, I haven't got around to that project yet, so I haven't much to offer.  Good luck with it, and let us know how it works out.

Best, Thom.

ftab, 2018-04-24:
+1. I've got the bytcht_nocodec module enabled in the kernel I'm using (also Ubuntu 16.04), and I've modprobed it, but I'm not quite sure how to proceed to tell it to use that driver. aplay -l shows nothing, or HDMI when i have that enabled... Any ideas?

tranfa, 2018-05-23:
Still unable to pass audio to the AudioGD I2s HDMI converter.

Just as a check: to which pin of the GPIO would you suggestion me to link the following pins of the AudioGD?

- Data
- Bck
- Wck
- Mck
- Ground
- Power

tranfa, 2018-06-08:
TheoRettisch thank you, this seems an interesting comment, albeit not sure how much hope this leaves me... :) 
What I'd like to do is to bring out of the Minnowboard the audio stream to an external DAC via I2S (through the I2S->HDMI Converter). 
I managed doing this with a Raspberry and audio results are good; I was hoping to do the same with even better results on an Intel based board like Minnowboard is.
In your answer, you are implying that I should add a piece of hardware (the Realtek one), if I understand correctly? Or could a piece of software (a driver) be enough? 
Thanks in advance for your answer.

TheoRettisch, 2018-06-13:
Hi Tranfa,

i dont think a driver will be enough to output audio via I2S. If you look at the elinux wiki here you will see this note: 
---------------
LPE Audio Support
NOTE: Audio is available via I2S, but requires a codec chip to be useful. This is intended to be resolved in a later lure.
---------------
https://elinux.org/Minnowboard:MaxBios

I have never seen such a lure. So i think the I2S ports are rather useless unless one possesses the skills to adapt an I2S codec.

Strange thing is the CPU has an onboard Azalia codec, but it seems that is only used for direct HDMI sound output. For my project i moved to an USB XMOS solution to output I2S sound to my DSP. With this solution i also don't use the potentially noisy clock from the Minnowboard which should help with sound quality.

ftab, 2018-07-27:
Let's say in theory we have a codec we're interested in developing a driver for, and we want a regular I2S 48k 24 bit stereo signal to come out of the Minnowboard's I2S port so when we test bringing up the codec from user space i2c or spi, we actually hear something. Is there any way that the "nocodec" driver is usable in that regard, or would we have to write the codec driver first and just hope it works? How do you then write and enable that machine driver on the Minnowboard so that Linux uses that as the audio output? Some kind of device tree file, or is it more complicated than that?

I have a feeling it's easier to "just" use some codec for which there's already a driver (such as a Realtek HDA) but still kind of in the dark as to how to (configure the board to?) instruct Linux to use it.

TheoRettisch, 2018-07-27:
I just had a look at the bytcht_nocodec module. After you loaded it did you check /proc/asound/cards is there is a new device "bytchtnocodec"?

I don't think there is an additional codec needed if you just want to use the I2S signal. But additional config in Alsa might be needed. I found this interesting repo, but not sure what no make out of it.

[](https://github.com/plbossart/UCM/tree/master/bytcht-nocodec)
[](https://github.com/plbossart/MinnowBoardMaxFirmware)

BetaMadMax, 2018-11-06:
> Hey guys, adding a codec on i2s isn't too complicated, at least on the
> hardware side. Just a bitclock, a frameclock, data in (assuming you want
> audio in), data out (assuming you want audio out). Linux has a huge list of
> codec drivers, but you might pick one that on a popular raspberry pi hat,
> like the Realtek ALC5236 Google put on their v2 voice kit bonnet, or the
> AD1938 of the Bela Face cape, or so many other options. Codecs also have a
> configuration interface like I2C but that's just two wires. On the sw side,
> you'll need to configure the I2S peripheral (linux kernel source dir
> linux/sound/soc/intel/baytrail/ ) and grab a driver
> (sound/soc/codecs/rt5645.c for example), then all the glue like device tree
> overlays and alsa config, etc. Adapting the face cape to be minnowboard
> Lure would be pretty cool. I'd kick out the kicad design if someone
> committed doing the driver work. Or you could use the cuttle(bone) lure to
> adapt the cape to the minnowboard.

Hey Paul, the turns that life gives and this world is so small ... I was
thinking about that same solution for a few weeks and I've even been in
talks with Henry of the Bela Project to achieve a Lure like that. The idea
is that it can be operated with, and without (just the codec or FACE Cape),
the SoC of the Beagle Bone, this is so in case we want to use the
middleware( based on Xionami) optimized for low latency and hardware
independent of the MinnowBoard. Communicate with me by MaxReyesR (at) Gmail
(dot) Com Maybe you want to collaborate with us in our project:
https://hackaday.io/project/7943-tabbx-20th-anniversary-bebox