MinnowBoard-org/bugs-and-help #78: Missing PWM in /sys/class/pwm

sabuffa, 2017-11-21:
Anybody can help with this?  I am running RHEL 7.4 with linux kernel 3.10 (the latest version of Red Hat), and cannot get the pwm to work.

It is missing in /sys/class/pwm and I'm not sure how to get it in there.  

I have the following:

/lib/modules/<linux_ver>/kernel/drivers/pwm/pwm-lpss.ko.xz

The output of modinfo pwm_lpss shows this:
>filename:       /lib/modules/3.10.0-693.el7.x86_64/kernel/drivers/pwm/pwm-lpss.ko.xz
alias:          platform:pwm-lpss
license:        GPL v2
author:         Mika Westerberg <mika.westerberg@linux.intel.com>
description:    PWM driver for Intel LPSS
rhelversion:    7.4
srcversion:     8335BF97ED3B26609FDB902
alias:          acpi*:80860F09:*
depends:
intree:         Y
vermagic:       3.10.0-693.el7.x86_64 SMP mod_unload modversions
signer:         Red Hat Enterprise Linux kernel signing key
sig_key:        4F:FD:D6:3C:93:7E:B4:A7:A1:14:BC:5E:89:1A:CB:DE:50:20:65:21
sig_hashalgo:   sha256

my boot kernel config shows:

> CONFIG_PWM=y
CONFIG_PWM_SYSFS=y
CONFIG_PWM_LPSS=m

What I don't have are these:

>PWM_LPSS_PCI
PWM_LPSS_PLATFORM

Any ideas how to get the pwm chip to show up for me?  I have successfully gotten the pin on the low speed header to be used as GPIO, however not as PWM.

gvancuts, 2017-11-22:
This is likely a driver missing in the kernel. I'm not very familiar with RHEL but that kernel is old and I'am assuming that RedHat does a lot of work back-porting new drivers over to that one to support various processors and devices. Looking at that table [here](https://access.redhat.com/support/policy/intel), I don't see the MinnowBoard processors ([Quad-Core Intel® Atom™ E3845](https://ark.intel.com/products/78475/Intel-Atom-Processor-E3845-2M-Cache-1_91-GHz) or [Dual-core Intel® Atom™ E3826](https://ark.intel.com/products/78477/Intel-Atom-Processor-E3826-1M-Cache-1_46-GHz)) on the list so you may be left with having to back-port the driver yourself. I would also advice you contact RedHat to confirm this with them.

sabuffa, 2017-11-22:
After more research it seems that the CONFIG_PWM_LPSS_PLATFORM is the important one that I need that I am missing.  According to the internet, this was added to linux kernel 3.18.  So I think you are correct that I would need to backport this to 3.10.  Since that's probably more trouble than its worth, I may instead attempt a software based pwm solution.  It is an order of magnitude slower but may still be fast enough for my needs.

gvancuts, 2017-11-23:
It may not be as hard as it sounds. I looked at it briefly today and was able to compile this driver in a kernel 3.10 tree without any effort. What I did was:

1. Copy those four files from a 3.18 kernel tree to the 3.10 kernel tree (put them under `drivers/pwm/`)

    ```
    pwm-lpss.h
    pwm-lpss.c
    pwm-lpss-platform.h
    pwm-lpss-platform.c
    ```

2. Modify the `drivers/pwm/Kconfig` file by adding the text below (make sure to put it **inside** the `if PWM` section

    ```
    config PWM_LPSS
            tristate "Intel LPSS PWM support"
            depends on X86
            help
              Generic PWM framework driver for Intel Low Power Subsystem PWM
              controller.
    
              To compile this driver as a module, choose M here: the module
              will be called pwm-lpss.
    
    config PWM_LPSS_PLATFORM
            tristate "Intel LPSS PWM platform driver"
            depends on PWM_LPSS && ACPI
            help
              The platform driver for Intel Low Power Subsystem PWM controller.
    
              To compile this driver as a module, choose M here: the module
              will be called pwm-lpss-platform.
    ```

3. Make sure the driver is selected in kernel config and rebuild it

I have not tested if this is fully functional but at least from a kernel internal API perspective, it did not complain and compiled successfully... at this stage, I'd give this a shot if I were you!

sabuffa, 2017-11-29:
It took me a few days but I was able to backport the driver to my version of RedHat and it seems to work.

closed 2017-12-21 by sabuffa